windytan/redsea #3: Compilation instructions incorrect for Linux

nivex, 2015-02-08:
Following instructions in readme:

    kjotte@daedalus:~/git/redsea$ gcc -o rtl_redsea -lm rtl_redsea.c
    rtl_redsea.c: In function ‘main’:
    rtl_redsea.c:91:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
         for (int iq=0;iq<=1;iq++) {
         ^
    rtl_redsea.c:91:5: note: use option -std=c99 or -std=gnu99 to compile your code

Code compiles correctly with command:
    kjotte@daedalus:~/git/redsea$ gcc -std=gnu99 -o rtl_redsea rtl_redsea.c -lm

parameter-pollution, 2015-02-08:
You added -std=gnu99, but you didn't move the "-lm" parameter to the end of the command as nivex did. If it's not at the end i get linker errors on my linux box.
Great tool btw.

windytan, 2015-02-08:
Right, fixed now.

closed 2015-02-09 by windytan