windytan/redsea #4: No output from redsea.pl

nivex, 2015-02-08:
rtl_redsea is emitting a sequence of 0's and 1's, but the Perl script doesn't seem to be decoding anything:
```
kjotte@daedalus:~/git/redsea$ ./rtl_redsea -f 90.7M | ./redsea.pl 
Found 1 device(s):
  0:  Realtek, RTL2841UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Name "main::correct_all" used only once: possible typo at ./redsea.pl line 151.
Found Rafael Micro R820T tuner
Tuner gain set to automatic.
[R82XX] PLL not locked!
Tuned to 312590 Hz.
Oversampling input by: 5x.
Oversampling output by: 1x.
Buffer size: 6.55ms
Exact sample rate is: 1250000.002070 Hz
Sampling at 1250000 S/s.
Output at 250000 Hz.
```

windytan, 2015-02-08:
You seem to be tuned to 312.5 kHz.

nivex, 2015-02-08:
eek! It didn't like the 'M' notation for some reason. I manually specified and it appears to be tuned correctly but still not seeing any data:

```
kjotte@daedalus:~/git/redsea$ ./rtl_redsea -f 90700000 | ./redsea.pl 
Found 1 device(s):
  0:  Realtek, RTL2841UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Name "main::correct_all" used only once: possible typo at ./redsea.pl line 151.
Found Rafael Micro R820T tuner
Tuner gain set to automatic.
Tuned to 91012500 Hz.
Oversampling input by: 5x.
Oversampling output by: 1x.
Buffer size: 6.55ms
Exact sample rate is: 1250000.002070 Hz
Sampling at 1250000 S/s.
Output at 250000 Hz.
```

windytan, 2015-02-08:
It will output bits at 1187.5 bps even if the bitstream doesn't make sense, i.e. the error correction is only done in the Perl script, and on a noisy channel the script won't print a single correct group. Maybe this could be corrected.

nivex, 2015-02-08:
I finally started getting some noisy data on the fourth station I tried.
Closed: operator error :)

closed 2015-02-08 by nivex