windytan/redsea #12: Incorrect frequency reported in display output

mvglasow, 2015-03-26:
```
$perl redsea.pl 99.0M
Waiting for sync at 99.00 MHz
Found 1 device(s):
  0:  Realtek, DVB-T Dongle, SN: 00000991

Using device 0: Dexatek DK DVB-T Dongle (Logilink VG0002A)
Found Fitipower FC0013 tuner
Tuner gain set to automatic.
Tuned to 99500000 Hz.
Oversampling input by: 8x.
Oversampling output by: 1x.
Buffer size: 4.10ms
Exact sample rate is: 2000000.052982 Hz
Sampling at 2000000 S/s.
Output at 250000 Hz.
```

As you can see, the "Tuned to" line reports a frequency that is 0.5 MHz above the selected one. It does seem to pick the correct frequency, though. (In my area I have two radio stations which are 0.5 MHz apart, and if I pick the lower one, I get the correct station.)

I didn't investigate further – it may be a bug in rtl-sdr.

mvglasow, 2015-03-26:
Same symptoms when I run rtl-fm directly:

```
$rtl_fm -f 99.0e6 -M wbfm -s 200000 -r 48000 - | aplay -r 48k -f S16_LE
Found 1 device(s):
  0:  Realtek, DVB-T Dongle, SN: 00000991

Using device 0: Dexatek DK DVB-T Dongle (Logilink VG0002A)
Found Fitipower FC0013 tuner
Tuner gain set to automatic.
Tuned to 99316000 Hz.
Oversampling input by: 6x.
Oversampling output by: 1x.
Buffer size: 6.83ms
Sampling at 1200000 S/s.
Output at 200000 Hz.
Playing raw data 'stdin' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
```

Looks like it's not redsea...

closed 2015-03-26 by mvglasow

windytan, 2015-03-27:
This is covered by rtl_fm's FAQ.

Q: Rtl_fm says it tunes to some other frequency instead of mine.

This is normal behavior, to dodge the DC spike present in E4000 tuners. The software tunes slightly off center and then corrects for the offset during demodulation.

mvglasow, 2015-03-27:
Thanks for the heads up... should've Read The Fine Manual :-)