windytan/redsea #13: Can't get it to decode anything

niston, 2016-05-06:
Maybe the hardware is not powerful enough?

pi@raspberrypi:~/redsea/src $ rtl_fm -M fm -f 88.2M -l 0 -A std -p 0 -s 228k -F 9 | ./redsea
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
HWTUNER Rafael Micro R820T
Tuner gain set to automatic.
Tuned to 88656000 Hz.
Oversampling input by: 8x.
Oversampling output by: 1x.
Buffer size: 4.49ms
SAMPLERATE 1824000.017395
Sampling at 1824000 S/s.
Output at 228000 Hz.
buffer underrun!

...is all Im getting.

windytan, 2016-05-09:
I haven't tested it on RPi yet, maybe soon. Thanks for the report.

kms, 2016-09-07:
There are some recent optimizations. I'm running it successfully on an RPi2 now, the rtl_fm and redsea processes are using about 40% CPU each. Might still be a tad too slow for the RPi1, but one option could be to direct the output from rtl_fm to a file and then post-process with redsea. No idea if that makes sense for your use-case, though.

windytan, 2016-09-16:
Works on RasPi 1 after recent optimizations. I also dropped the sampling rate from 228k -> 171k and added a tip in the readme about fast integer approximation in the FM decoder.

closed 2016-09-16 by windytan