windytan/redsea #16: redsea: data.cc:10: std::string redsea::getLCDchar(int): Assertion `code >= 32' failed.

martinzsmith, 2016-07-06:
I run readsea with the below command, any ideas what I'm doing wrong ?
$ date ; rtl_fm -M fm -f 88.5M -l 0 -A std -p 0 -s 228k -F 9 | ./redsea 2>&1  > /tmp/ram/redsea_88.5M.log.B.txt ; date
Tue Jul  5 13:02:27 UTC 2016
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to automatic.
Tuned to 88956000 Hz.
Oversampling input by: 8x.
Oversampling output by: 1x.
Buffer size: 4.49ms
Exact sample rate is: 1824000.017395 Hz
Sampling at 1824000 S/s.
Output at 228000 Hz.
redsea: data.cc:10: std::string redsea::getLCDchar(int): Assertion `code >= 32' failed.
Signal caught, exiting!

User cancel, exiting...
Aborted
Tue Jul  5 14:09:12 UTC 2016
$
---
It looks to happen at random times:
```
-rw-r--r--  1 mzs mzs   23356 Jun 30 03:50 redsea_88.5M.log.0.txt.xz
-rw-r--r--  1 mzs mzs    3764 Jun 30 10:12 redsea_88.5M.log.1.txt.xz
-rw-r--r--  1 mzs mzs    3208 Jul  1 01:19 redsea_88.5M.log.2.txt.xz
-rw-r--r--  1 mzs mzs    7864 Jul  1 17:29 redsea_88.5M.log.3.txt.xz
-rw-r--r--  1 mzs mzs    6364 Jul  1 22:57 redsea_88.5M.log.4.txt.xz
-rw-r--r--  1 mzs mzs   12092 Jul  2 03:36 redsea_88.5M.log.5.txt.xz
-rw-r--r--  1 mzs mzs    9040 Jul  2 10:46 redsea_88.5M.log.6.txt.xz
-rw-r--r--  1 mzs mzs   34284 Jul  2 22:57 redsea_88.5M.log.7.txt.xz
-rw-r--r--  1 mzs mzs   29644 Jul  3 12:02 redsea_88.5M.log.8.txt.xz
-rw-r--r--  1 mzs mzs   31604 Jul  4 00:14 redsea_88.5M.log.9.txt.xz
-rw-r--r--  1 mzs mzs   11460 Jul  5 04:00 redsea_88.5M.log.A.txt.xz
-rw-r--r--  1 mzs mzs    2136 Jul  5 14:09 redsea_88.5M.log.B.txt.xz

```

windytan, 2016-07-06:
Hi! Thanks for your extensive testing!

You were doing nothing wrong; this is a legit bug. It should be fixed now.

closed 2016-07-06 by windytan

martinzsmith, 2016-07-06:
Thanks for the fix, I think that the root of the problem is my reception, sometimes a bit is flipped the wrong way. See below:
```
$ xzcat *.xz | grep "ps" | grep -v alt_freqs | sort | uniq -c | sort -rn
    340 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RTE R1  "}
      4 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":""}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"VVE R1  "}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RTGŞR1  "}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RTE Zå  "}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RTE R1sæ"}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RTE R1e"}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RTE R1ª‖"}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RTE R1 Å"}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RTE R1,%"}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RTE R;  "}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RTE Qš  "}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RTE q2  "}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RTěǦR1  "}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RTB0R1  "}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RTAWR1  "}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"RT æR1  "}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"OæE R1  "}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"Õ4E R1  "}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"bTE R1  "}
      1 {"pi":"0x2201","group":"0A","tp":"true","prog_type":"Varied Speech","ta":"false","ps":"BRE R1  "}
$

```
Your original code would work perfectly if only my RF reception was better :)

windytan, 2016-07-06:
I think these could be mitigated by improving the error detection & correction (I've made an issue about it) and also implementing clock phase recovery, in the future.