windytan/redsea #17: [liquid.h] building error

iDoka, 2016-07-27:
By building I got:
```
$ make
make  all-recursive
make[1]: Entering directory `/home/iDoka/soft/redsea'
Making all in src
make[2]: Entering directory `/home/iDoka/soft/redsea/src'
g++ -std=gnu++11 -DHAVE_CONFIG_H -I. -I..  -g -Wall -Wextra -Weffc++ -Wstrict-overflow -Wshadow -Wuninitialized -pedantic    -Wall -std=c++11 -MT redsea-redsea.o -MD -MP -MF .deps/redsea-redsea.Tpo -c -o redsea-redsea.o `test -f 'redsea.cc' || echo './'`redsea.cc
In file included from mpx2bits.h:11:0,
                 from bits2blocks.h:6,
                 from redsea.cc:24:
liquid_wrappers.h:6:27: fatal error: liquid/liquid.h: No such file or directory
 #include "liquid/liquid.h"
                           ^
compilation terminated.
make[2]: *** [redsea-redsea.o] Error 1
```

I try to search in repo, but cant find proper package:
```
$ yum search liquid
=================== N/S matched: liquid ===================
rubygem-liquid-doc.noarch : Documentation for rubygem-liquid
rubygem-liquid.noarch : A secure, non-evaling end user template engine with aesthetic markup
```

My system is CentOS7 x86_64

iDoka, 2016-07-27:
thanks for explanation of prerequisites
Now I success install redsea

closed 2016-07-27 by iDoka

windytan, 2016-07-29:
Hi, yes, it needs liquid-dsp and I've added this more clearly in the Readme.