windytan/redsea #24: broken json

flux242, 2016-09-17:
{"pi":"0xd318","group":"3A","tp":"true","prog_type":"Pop Music","open_data_app":{"oda_group":"12A","app_name":"RadioText+ (RT+)"}

there's a missing }

windytan, 2016-09-17:
Thanks, fixed.

closed 2016-09-17 by windytan

flux242, 2016-09-17:
you may test the json output using jq command. Example to show the tmc events -
`nc -u -l 192.168.1.255 6666 | ./redsea -b | sed -r 's/\/\*[^\/]+.//g'|jq -C --unbuffered -c 'select(.group=="8A")|.tmc.message|select(.location!=null)|[.location,.event_codes,.description]' | awk '{if(old!=$0)print $0;old=$0}'`

the sed filter is needed because jq doesn't like comments in the json (like /* TODO */). The awk at the end removes doubles. jq in this example will of course parse the rest and will produce an error if the json is not valid.

windytan, 2016-09-17:
Great tip! I tried jsonlint but it had to be called separately for each line of JSON, and thus there was a lot of overhead. (Just changed `-r` to `-E` for BSD `sed`)

flux242, 2016-09-17:
you are welcome! Thanks for the great program