windytan/redsea #26: tmc description is missing

flux242, 2016-09-17:
the following is selected: [.location,.event_codes,.description]'

result:
[12823,[243],"."]
[12823,[243],"."]
[12820,[501],"."]
[31769,[802],"."]
[61452,[401,1472],"."]
[12744,[1926],"."]
[7613,[105],"."]
[7613,[105],"."]
[32245,[802,810],"."]
[7613,[241],"."]
[7613,[241],"."]
^C

flux242, 2016-09-17:
ah, the csv files aren't linked as resources to the executable so the cwd should contain the data folder

windytan, 2016-09-17:
Thanks for the report, is this something I could improve on in how the repository is set up?

flux242, 2016-09-17:
well, the csv files in the data folder could be linked to the executable. In this case it won't be necessary to have data folder in the current working directory (cwd).

windytan, 2016-09-17:
Reopening this as a reminder.

windytan, 2016-09-20:
fixed v0.7.3

closed 2016-09-20 by windytan