windytan/redsea #27: item.artist and item.title are missing last character

flux242, 2016-09-17:
I know this part is still in the development, just wanted to point this out:
`$ nc -u -l 192.168.1.255 6666 | ./redsea -b | sed -r 's/\/\*[^\/]+.//g'|jq -c 'select(.group=="12A")|.radiotext_plus|select(.["item.artist"]!=null)|[ .["item.artist"],.["item.title"] ]'
["            ","            "]
["            ","            "]
["            ","            "]
["            ","            "]
[" Taylor Swif"," Shake It Of"]
`
it should be "Taylor Swift" I suppose and "Shake It Off"

windytan, 2016-09-17:
I can't reproduce this; it was fixed already in commit [55c96fa](https://github.com/windytan/redsea/commit/55c96fa5163f2751f8c1c88a14d433f4abef2bb1). Have you `git pull`ed the latest version?

flux242, 2016-09-17:
it is not always happen. Check the output of the command above:
["                         ","            "]
["Robin Schulz & J.U.D.G.E.","Show Me Love"]
["                ","                "]
["Enrique Iglesias","Duele El Corazó"]
["              ","                   "]
["Meghan Trainor","All About That Bass"]
["                              ","                   "]
["Puff Daddy & Faith Evans & 112","I'll be missing you"]
["         ","           "]
["Elle King","Ex's & Oh's"]
["              ","                     "]
["Ellie Goulding","Still Falling For You"]
["          ","              "]
[" The Rasmu"," In the shadow"]
["                       ","              "]
["Kygo feat. Parson James","Stole the Show"]
["              ","               "]
["Jennifer Lopez","Ain't Your Mama"]
["     ","                "]
["Adele","Someone Like You"]
["               ","                       "]
["Marlon Roudette","When The Beat Drops Out"]
["    ","                  "]
["Asia","Heat of the moment"]
["                           ","         "]
["Mike Perry Feat. Shy Martin","The Ocean"]
["            ","        "]
["Shawn Mendes","Stitches"]

only the '[" The Rasmu"," In the shadow"]' is wrong
Let the command run some time. If this will not help I could also record the input of the redsea so you could pipe it directly with 'cat'

flux242, 2016-09-17:
and yes I have updated the master today

windytan, 2016-09-17:
Interesting. I don't have any RT+ stations here, just a set of test data from Germany but all of it seems to decode normally. A recording in the form of hex output (`-x`) would be helpful.

flux242, 2016-09-17:
ok, I need to go now but I will record the stream today night or tomorrow

flux242, 2016-09-17:
ok, here is the dump file https://drive.google.com/file/d/0B_LP1ADmClSdRGlDWmZqdWM2SGs/view?usp=sharing

file contains ASCII '0' and '1' characters to be compatible with the redsea -b option

there's the bayern1 radio station at the beginning and then Antenne Bayern.
 Usage:
`uncompress -c radio.txt.Z | ./redsea -b` 

uncompress is installed with the 'ncompress' package - `sudo apt-get install ncompress`

tell me if you need more samples like this

windytan, 2016-09-17:
The problem seemed to be in how `string::length()` and `string::substr()` count UTF-8 characters. Fixed now. (There remains another bug where RT+ length changes are not entirely in sync with RT changes)

closed 2016-09-17 by windytan

flux242, 2016-09-18:
with the change number f69e763ae27fa74b72f6f the output looks like:
["                      ","       "]
["Felix Jaehn feat. ALMA","Bonfire"]
["              ","        "]
["Party Mix 1692","Mix 1692"]
["                ","                "]
["Enrique Iglesias","Duele El Corazó"]
["              ","        "]
["Party Mix 1883","Mix 1883"]
["                 ","                "]
["Christina Stürme"," Ein Teil von mi"]
["              ","        "]
["Party Mix 1512","Mix 1512"]

and with the change number d166dc852db28f3a it looks like this:
["Felix Jaehn feat. ALMA","Bonfire"]
["ehn feat. ALMA","Bonfire "]
["Party Mix 1692","Mix 1692"]
["h ? 0800 - 994 1","                "]
["Enrique Iglesias","Duele El Corazón"]
["tenne.de      ","        "]
["Party Mix 1883","Mix 1883"]
["it: digital via D","S (Astra)       "]
["Christina Stürmer","Ein Teil von mir"]
["Stürmer - Ein ","l von mi"]
["Party Mix 1512","Mix 1512"]

is this what you mean with the remaining bug? Looks like RT text?

windytan, 2016-09-18:
Yes; this is issue #23

flux242, 2016-09-18:
but rt+ tags seems also broken to me
["Christina Stürmer","Ein Teil von mir"]
["Stürmer - Ein ","l von mi"]  <-- here it is broken

windytan, 2016-09-18:
"Stürmer - Ein " is 14 characters; the length is erroneously applied from the next tag:
* `Stürmer - Ein `
* `Party Mix 1512`

Same with "l von mi":

* `l von mi`
* `Mix 1512`

The erroneus RT+ tags are only shown for a short time compared to the correct ones. Will need to think about how to detect this.

flux242, 2016-09-18:
ah, ok, I see.

windytan, 2016-09-18:
Most cases fixed now.

flux242, 2016-10-11:
btw:

std::string rtrim(std::string s) {
  int last_non_space = 0;
  for (size_t i=0; i<s.length(); i++)
    if (s.at(i) != ' ')
      last_non_space = i+1;
  return s.substr(0,last_non_space);
}

can be replaced with
std::string rtrim(std::string s) {
  return s.erase(s.find_last_not_of(' ')+1);
}
note that a string copy is created.

Or without a copy created
std::string & rtrim(std::string& s) {
  return s.erase(s.find_last_not_of(' ')+1);
}

windytan, 2016-10-11:
Thanks! I used to have something like that but it used `std::not1` which is a C++14 feature. That one seems to work in C++11 as well.