windytan/redsea #30: Noise issues

windytan, 2016-09-30:
I've gotten some reports that redsea requires a stronger FM signal than other decoders. Possible reasons discussed here.

## PLL

The PLL (realized as `nco_crcf_pll_step` in liquid-pll) takes some time to lock onto the subcarrier. What could affect this? Loop filter parameters? (Time in seconds here.)

[image]

mvglasow, 2017-04-25:
My setup’s slightly different, but based on the same code and thus likely to suffer from the same issues: audio samples are obtained with C code based on `rtl_fm`, which are then processed into a RDS binary stream using a Java implementation based on an early (ca. April 2015) version of redsea. This is implemented as a plugin to RDS Surveyor, which does all the higher-level RDS stuff, in addition to visualizing signal strength and RDS block error rates. The plugin also downsamples audio to 48,000 Hz and plays it through the sound card—helpful because the amount of static is a rough indicator for signal quality.

I use a Logitech VG0002A (FC0013 tuner), no extra shielding, hooked up to my roof antenna. OS is Ubuntu MATE 16.04 (64-bit), running on an Intel Core2 Duo P8400 @ 2.26GHz × 2. For reference, I also use a Si4703-based dongle, i.e. a dedicated RDS FM tuner, wrapped in aluminium foil for shielding, with the same software setup.

I notice the Si4703 picks up fewer stations than the RTL2832, though RDS is usually crisp with hardly any block errors. With the RTL2832 occasionally I get good RDS data, other days things look quite messy. The antenna setup in my house probably isn’t the greatest either, I am also having issues with analog radio reception.

My observations:
* Block errors kick in once CPU load is constantly above 80% on both cores; at 128K without audio load is usually around 60–80% on both cores. Enabling audio eats a few extra CPU cycles; disabling it slightly improves RDS reception. 
* Sample rate matters. At 128K things look decent but going up as far as 250K improved signal quality (this was before I implemented audio). With more CPU power, you might be able to go higher.
* The atan algorithm in `rtl_fm` (`-A` command line option) makes a difference. `std` and `fast` seemed almost identical in terms of signal quality and CPU load, but with `lut` I immediately noticed extra static and didn’t get a single RDS group.
* Automatic gain control on the FC0013 seems botched. On some occasions, I’d get the strongest local station at 20+ dBm, on others it was somewhere around -5. This changed from time to time for no apparent reason—I suspect gain control to be the culprit. I’ve worked around this by controlling gain in software. This was implemented only today, hence no long-term test results yet, but so far gain has consistently gone up to the maximum of 197 dB.
* It seems the system needs some time to stabilize. On startup, I frequently see patterns of good and bad blocks alternating quite reproducibly, then the error rate gradually improves. I’m wondering if this could be some kind of timing issue.
* Disabling `rtl_fm` resampling resulted in signal strength improving by some 5 dBm.

windytan, 2017-04-25:
Thanks, great observations!
Note that the current versions, based on liquid-dsp, are much more noise-resistant and efficient; CPU load is at 0.8 % on my 2.8 GHz Intel Core i7.

mvglasow, 2017-05-01:
Thanks for the heads-up; this is where my implementation differs, as I needed the carrier demodulation part to be in pure Java and thus used [Java DSP Collection](http://www.source-code.biz/dsp/java/). It’s a lot less feature-complete than liquid-dsp; modernizing that part of the code will take some more research.

I just enabled csv stats on my implementation and monitored the PLL frequency. On a good sample, it jumps about wildly for about a quarter of a second, then goes to 57002 and slowly decreases to around 56997.5 with minimal oscillation, though that takes a few seconds.

Sometimes, after things have stabilized, I have brief periods of mostly bad samples, or even temporary interruptions, which quickly return to normal. These seem to correspond to the frequency diverging from its stable value and then returning in the stats.

I’m wondering if the pilot tone is subject to similar jitter. If the pilot tone remains stable while the PLL frequency is all over the place, it may be a sign that the PLL is doing weird things (or something’s specifically messing up the frequency range of the subcarrier). If there is similar jitter in the pilot tone—I’d expect the changes in the PLL frequency to slightly lag behind—we may have some issue that affects all frequencies and the PLL jitter is just a symptom of it. Sporadically dropped samples come to mind, or maybe jitter in the sample rate.

In the latter case, it would seem logical to rely on the pilot tone as a tuning reference. However, I see that you dropped pilot tone recovery in e4edd12—what was the motivation for that? Did you observe any changes in noise resilience before and after?

windytan, 2017-05-01:
Pilot tone recovery was dropped because not all RDS-carrying stations are guaranteed to have a pilot tone. Such a station is, for instance, a local station here that is monaural and thus has no stereo subcarrier or pilot tone, yet transmits a PS name via RDS.

It could be possible to detect the presence of a pilot tone and base our selection of clock base on that; or use a command-line switch. As per the RDS standard, the RDS subcarrier _is_ supposed to be locked to the third harmonic.

mvglasow, 2017-05-02:
Interesting—in fact I was wondering the other day if the pilot tone is present on monaural stations which transmit RDS.

Another thing I noticed here: when I turn the dial and pick up some noise before tuning into a good station, it takes a very long time before I get some RDS data. In one instance, I spent some 2–3 minutes listening to static before turning into the strongest local station here, no RDS until I gave up 5 minutes later. Analysis showed that for most of the latter part, the PLL remained stubbornly locked to some 56600 Hz, with remarkable stability.

Conclusion: Noise can profoundly throw the PLL off, an effect which seems to increase with the duration of the noise, and it can take a long time to recover even when a good signal is received again.

Another approach might be to run an FFT on the baseband signal and look for the pattern of the RDS subcarrier around 57 kHz—we should see two peaks, some 1.65 kHz apart, and a local minimum in the middle, which would be the subcarrier frequency. If we don’t see a clear RDS subcarrier pattern, we’re probably listening to a non-RDS station (or even plain static) and should leave the subcarrier frequency alone.

While we’re at it, we could check for both the pilot tone and the subcarrier and, if both are present, see if they agree. Since the signal level of the pilot tone is higher than that of the RDS subcarrier (10% vs. 5%), I’d expect the pilot tone to have better noise resilience.

mvglasow, 2017-05-05:
Results are discouraging.

I gave up on pilot tone recovery, as it saturated my CPU, negating any potential improvements in reliability. Maybe redsea lends itself better to that approach.

The DFT approach doesn’t seem to work—unless I’ve made an error somewhere. I collected two seconds of samples (which should give me 0.5-Hz frequency bins), then ran a DFT at 17/19/21 kHz as well as at 54/56.175/57/57.825 kHz and looked for the peak pattern.

No matter whether I was listening to a good station or static, the DFT analysis kept flipping happily between detecting and losing the pilot tone and/or the RDS subcarrier (independently of each other). Frequently, on the good station, DFT analysis would tell me it had lost the subcarrier but the decoder would happily continue spitting out data.

mvglasow, 2017-05-09:
A few more ideas:

I see the signal level of the good MPX looks more “balanced” than that of the noisy one, which looks more jagged, with occasional peaks. Possibly as a result of those, AGC gain is lower. I wonder if it’s possible to set the AGC to a more “aggressive” setting (higher gain) and how this would affect the PLL and everything after.

Speaking of gain control—when I mentioned gain control in software, I meant “determine in software what gain value to set the tuner to”. For redsea, that would mean controlling `rtl_fm` parameters.

Frequency correction in the “homebrew” PLL looked like this:
`fsc -= 0.5 * pll_beta * d_phi_sc;`
I’m by no means an expert on signals, but would it be possible to factor signal quality into that equation, as a kind of confidence level, which would be lower when noise is detected? (And would this still be possible with liquid-dsp?)

As for detecting noise, I just worked something out for an FM seek algorithm and am quite happy with the results: Because RSSI is a poor indicator of a valid FM station (there are weak transmitters as well as strong noise), I analyzed the demodulated spectrum up to the RDS subcarrier frequency. I ran an FFT across the spectrum, then determined the average power level (in dB) and mean average deviation. For noise, mean average deviation is lower than for a good station but the average power level is higher. By experimenting with some good and some bad signals I established a threshold for both values, then did a simple weighted difference, with weights chosen so that the result would be zero for the threshold. The greater the value, the better the signal. Maybe that’s a basis for a confidence level.

To work around issues with noise detuning the PLL beyond recovery, I ended up monitoring the PLL frequency and allowing it to drift within +/-14 Hz of the subcarrier frequency. (7 Hz is the tolerance as per the specs, which I doubled to allow for similar inaccuracy on the receiving end). When the frequency goes outside that range, I simply reset it to 57 kHz—admittedly not the most elegant solution, but it dramatically improved behavior in scanning scenarios, altering between good stations, noisy stations and pure static.

windytan, 2017-05-10:
PLL drift/detuning is no longer a problem, now that liquid is in use. The only remaining issue there is jitter with very noisy signals, as in the pictures above, but I'm not even sure if that causes any degradation in quality.

As for the homebrewn PLL design in very old versions of redsea, I'm kind of reluctant to comment on that with no background in theory myself.

windytan, 2017-05-12:
After testing the current version of redsea against RDS Spy on a noisy MPX signal, it seems redsea performed better, recoving around twice as many groups. So I guess noise should not be a problem any more.

closed 2017-05-12 by windytan

andimik, 2025-12-05:
I have asked AI.

> GQRX's code uses filtering, PLLs, and demodulation specific to RDS at 57 kHz.
    These algorithms are potentially portable, but require adaptation from GNU Radio blocks to native C/C++ code.
    In conclusion: Yes, the core ideas and algorithms from GQRX can be implemented in Redsea, but it involves a non-trivial porting effort—mainly reimplementing GNU Radio blocks into C/C++.

This is an approach by AI (note, it's a draft and the sample rate is fixed 192k)

```
#include <math.h>
#include <complex.h>
#include <stdlib.h>
#include <string.h>

#define SAMPLE_RATE 192000.0
#define RDS_FREQ 57000.0
#define RDS_BANDWIDTH 2200.0
#define FILTER_TAPS 101

// Placeholder for filter coefficients (should be designed offline)
double fir_coeffs[FILTER_TAPS];

// PLL state
typedef struct {
    double phase;
    double freq;
    double loop_filter_state;
} PLLState;

// Initialize PLL
void init_pll(PLLState *pll) {
    pll->phase = 0.0;
    pll->freq = 0.0;
    pll->loop_filter_state = 0.0;
}

// Apply FIR filter (simple convolution)
void apply_fir(const float *input, float *output, int length, const double *coeffs, int num_taps) {
    for (int i = 0; i < length; i++) {
        double acc = 0.0;
        for (int j = 0; j < num_taps; j++) {
            if (i - j >= 0)
                acc += input[i - j] * coeffs[j];
        }
        output[i] = (float)acc;
    }
}

// Update PLL based on phase error
void update_pll(PLLState *pll, double phase_error, double dt) {
    // Loop filter (simple proportional-integral)
    double k_p = 0.1;
    double k_i = 0.01;

    pll->loop_filter_state += k_i * phase_error * dt;
    pll->freq = k_p * phase_error + pll->loop_filter_state;

    // Update phase
    pll->phase += (2 * M_PI * (RDS_FREQ + pll->freq)) * dt;
    if (pll->phase > 2 * M_PI)
        pll->phase -= 2 * M_PI;
    if (pll->phase < 0)
        pll->phase += 2 * M_PI;
}

// Generate local oscillator sample
static inline complex float generate_lo(PLLState *pll, double t) {
    double phase = pll->phase;
    return cosf(phase) + I * sinf(phase);
}

// Demodulate (quadrature detection)
float demodulate(const complex float sample, const complex float lo) {
    complex float mixed = sample * conjf(lo);
    return crealf(mixed); // Real part gives phase difference
}

// Main processing function
void process_rds(const float *input_samples, int num_samples) {
    static double fir_coeffs[FILTER_TAPS];
    static int initialized = 0;
    if (!initialized) {
        // TODO: Design and assign actual FIR coefficients here
        for (int i = 0; i < FILTER_TAPS; i++) {
            fir_coeffs[i] = 0.0; // placeholder
        }
        initialized = 1;
    }

    float filtered_samples[num_samples];
    apply_fir(input_samples, filtered_samples, num_samples, fir_coeffs, FILTER_TAPS);

    PLLState pll;
    init_pll(&pll);

    for (int i = 0; i < num_samples; i++) {
        double t = (double)i / SAMPLE_RATE;

        // Generate local oscillator
        complex float lo = generate_lo(&pll, t);

        // Frequency translation
        complex float sample = filtered_samples[i] + 0.0f * I; // Convert to complex
        complex float shifted = sample * conjf(lo);

        // PLL phase update
        double phase_error = atan2f(crealf(shifted), cimagf(shifted));
        update_pll(&pll, phase_error, 1.0 / SAMPLE_RATE);

        // Demodulate
        float bit_value = demodulate(shifted, lo);

        // Here, implement Manchester decoding and frame extraction
        // For illustration, just print the demodulated value
        printf("%f\n", bit_value);
    }
}
```

vladisslav2011, 2025-12-06:
> Could you share your samples from the screenshot?

MPX files, 192 kHz, 16 bit, single channel:

Goal: PI, full PS and partial RT
[gqrx_20230625_044401_91200000.wav](https://github.com/user-attachments/files/23972909/gqrx_20230625_044401_91200000.wav)
Goal: PI, full PS
[gqrx_20230625_044401_89500000.wav](https://github.com/user-attachments/files/23972911/gqrx_20230625_044401_89500000.wav)
Goal: PI, full PS and full RT
[gqrx_20230625_044401_107600000.wav](https://github.com/user-attachments/files/23972910/gqrx_20230625_044401_107600000.wav)
Goal: PI, full PS and partial RT
[gqrx_20230625_044401_98700000.wav](https://github.com/user-attachments/files/23972912/gqrx_20230625_044401_98700000.wav)

These are easy, but I have many more noisy samples, which RDS Spy fails to process and my Gqrx fork requires several passes to detect PI code. There are some "out of spec" samples too with strong RDS subcarrier at 56kHz or so and similar symbol rate mismatch, that cant be easily decoded even by my fork of Gqrx, but may be decoded with full PS after small source code adjustments. I can share these too.

> Apache license of Gqrx seems compatible.

Only some minor GUI components of Gqrx are licensed under Apache 2.0. All other parts of Gqrx, including the DSP code are licensed under GPL 3.0, that is not compatible with MIT, so direct copy-pasting is not possible. My RDS decoder implementation is just an unfinished experiment, not something production ready. Symbol/phase synchronizer still needs some work as it does not work reliably at low SNR <0.5 ... 0.6dB. Another thing to implement/test is belief-propagation style soft symbol ECC decoder... And, of course, major code cleanup is required as current RDS decoder is a bit messy.

> This is an approach by AI

Incomplete and non-functional as expected... No AI-generated code, please 😄

vladisslav2011, 2025-12-06:
> The noisy example WAVs will be very helpful

PI detection only:
[gqrx_20241116_004428_101500000.wav](https://github.com/user-attachments/files/23985444/gqrx_20241116_004428_101500000.wav)
[gqrx_20230625_045012_87900000.wav](https://github.com/user-attachments/files/23986162/gqrx_20230625_045012_87900000.wav)

> And I'd be very interested in the out-of-spec samples

Full PS
[gqrx_20240616_135025_95000000.wav](https://github.com/user-attachments/files/23986016/gqrx_20240616_135025_95000000.wav)

Partial PS and partial RT
[gqrx_20240616_135923_92501000.wav](https://github.com/user-attachments/files/23986017/gqrx_20240616_135923_92501000.wav)