windytan/redsea #32: RDS 2.0

SopaXorzTaker, 2016-10-16:
Soon, the new version of RDS standard will start to be integrated. It's backwards-compatible, but adds additional subcarriers to increase data throughput. It will have many new features: https://en.wikipedia.org/wiki/Radio_Data_System#RDS_2.0.

Will you implement it when the specifications are released?

windytan, 2016-10-16:
As soon as I get example data!

windytan, 2017-05-13:
Will reopen when at least one station transmitting RDS 2.0 can be received.

andimik, 2017-06-29:
Seems German public broadcaster NDR is testing RDS2.0, see http://radioforum.foren.mysnip.de/read.php?8773,1427903 

I've asked the user to get in contact with you.

andimik, 2022-05-07:
There is now one station in Italy using RDS2.0, see https://www.rundfunkforum.de/viewtopic.php?p=1655588#p1655588 (incl. mpx recording).

The 66.5 kHz carrier transmits the station logo (says the encoder manufacturer) 

The normal RDS is
```
$ redsea -r 192000 -f 53c5_rds2.wav 
{"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"dynamic_pty":true},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"group":"2A","pi":"0x53C5","prog_type":"Pop music","tp":true}
{"group":"2A","pi":"0x53C5","prog_type":"Pop music","tp":true}
{"group":"14A","other_network":{"pi":"0x53C5","tp":true},"pi":"0x53C5","prog_type":"Pop music","tp":true}
{"group":"3A","open_data_app":{"app_name":"RadioText+ (RT+)","oda_group":"12A"},"pi":"0x53C5","prog_type":"Pop music","tp":true}
{"country":"it","group":"1A","has_linkage":false,"pi":"0x53C5","prog_type":"Pop music","tp":true}
{"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"dynamic_pty":true},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ps":"GIOCONDA","ta":false,"tp":true}
{"group":"2A","pi":"0x53C5","prog_type":"Pop music","tp":true}
{"group":"2A","pi":"0x53C5","prog_type":"Pop music","tp":true}
{"debug":["TODO 10A"],"group":"10A","pi":"0x53C5","prog_type":"Pop music","tp":true}
{"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"clock_time":"2022-05-06T18:10:00Z","group":"4A","pi":"0x53C5","prog_type":"Pop music","tp":true}
{"clock_time":"2022-05-05T18:10:00-01:00","group":"4A","pi":"0x53C5","prog_type":"Pop music","tp":true}
{"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"dynamic_pty":true},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ps":"GIOCONDA","ta":false,"tp":true}
{"group":"2A","pi":"0x53C5","prog_type":"Pop music","tp":true}
{"group":"2A","pi":"0x53C5","prog_type":"Pop music","tp":true}
{"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"dynamic_pty":true},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ps":"GIOCONDA","ta":false,"tp":true}
{"group":"2A","pi":"0x53C5","prog_type":"Pop music","tp":true}
{"group":"12A","pi":"0x53C5","prog_type":"Pop music","radiotext_plus":{"item_running":true,"item_toggle":0},"tp":true}
{"group":"2A","pi":"0x53C5","prog_type":"Pop music","tp":true}
{"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"dynamic_pty":true},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ps":"GIOCONDA","ta":false,"tp":true}
{"group":"2A","pi":"0x53C5","prog_type":"Pop music","tp":true}
{"group":"2A","pi":"0x53C5","prog_type":"Pop music","tp":true}
{"group":"14A","other_network":{"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true},"pi":"0x53C5","prog_type":"Pop music","tp":true}
{"group":"1A","has_linkage":false,"language":"Unknown","pi":"0x53C5","prog_type":"Pop music","tp":true}
{"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
{"group":"3A","open_data_app":{"app_name":"RadioText+ (RT+)","oda_group":"12A"},"pi":"0x53C5","prog_type":"Pop music","tp":true}
{"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x53C5","prog_type":"Pop music","ta":false,"tp":true}
```

andimik, 2022-05-08:
https://www.rundfunkforum.de/download/file.php?id=10485&mode=view

This is the spectrum and a first try with Rdsspy and its new RDS2 extension.

andimik, 2023-08-29:
I hope I can record an MPX when I will be on the Adriatic sea in September.

andimik, 2023-09-16:
This is an (error-free) MPX recording of that station recorded near Udine (UD), Italy on 98.5 MHz from Transmitter Porzus.

Seems one cycle is around 1 min, so I've cut my initially 10 min recording to 4 min.

Please note the timestamp is not the original recording time.

[gioconda_20230915_udine_192khz_mono_4min.txt](https://github.com/windytan/redsea/files/12640098/gioconda_20230915_udine_192khz_mono_4min.txt)

[https://we.tl/t-njT2T4kkGp](https://we.tl/t-njT2T4kkGp?utm_campaign=TRN_TDL_05&utm_source=sendgrid&utm_medium=email&trk=TRN_TDL_05)

I found out that the broadcaster is using this kind of RDS encoder: https://pira.cz/rds/p164man.pdf

The author of many RDS hardware and also RDSSpy (Jan) wrote to me,

>    the RDS2 is dead, there's no support at all in the receivers, moreover
>    the station logo has been unexpectedly removed from final standard.
>
>    That Italian station uses some of the PIRA RDS encoder, they're sending
>    logo on the additional carrier but this has no effect at all. That
>    function has been implemented for potential future use but now I'm
>    almost sure that there's no future related to this.

Althought in this document from 2016, the RDS Forum still wrote about Logo implementation up to a size of 12 kB.

https://www.rds.org.uk/2010/pdf/R16_055_1.pdf

And in one of the latest publications on https://www.rds.org.uk/2010/pdf/Position%20paper%2023_01%20RDS2%20-%20Frequently%20Asked%20Questions.pdf the RDS Forum are still mentioning it.

There is a preview of the (very expensive) PDF: https://www.techstreet.com/products/preview/2566513

Annex E (page 33) seems to describe it anyway.

windytan, 2023-09-16:
Awesome research!

It's sad that RDS2 never took off. It would be nice to implement in redsea anyway, possibly, if someone ever wants to take a look at these rare data streams.

I'm just wondering how the output would look like. Pictures aren't very suitable for JSON printout. Maybe the user could give a special command-line switch to enable RDS2, and then another switch to point to a directory where to save these files?

sm3ulc, 2023-09-16:
Hi!

The pics will be very small so probably not much problem in having them and
other binary stuff encoded in BASE64. RDS2 har great potential but was
sadly much too late to the market.

Kind regards
// David

andimik, 2023-09-16:
[image]

This is the picture, 8.45 kB.

veso266, 2023-09-16:
Why couldn't RDS2 take off

The only encoder that I know supports is is Jans Pira, maybe we should start loobying with Hans to add it into his Stereo Tool

as for a decoder, even RDSSpy doesn't seam to support it (at least the public free version doesn't), I saw a video once, but thats it

as for receivers, well if almost no station is using it, no wonder no receiver will support it

This needs to be somehow changed (I don't know how, but I would realy loved for RDS2 to take off), RDS2 is way better then DAB, it gives FM new life, it adds extra goodies that enhance the expirience (I know a lot of people will disagree with me and I will loose what I love anyway, but why can't audio be transmitted to out ears in analog way.... whyy)

I mean if people adopted DAB+ why cannot RDS2 have the same future?

If anyone wants the standard, I have it and can send it to anyone that asks for it

andimik, 2023-09-17:
Please note that some countries will need the extended features of RDS2 (character sets, new lower frequency).

Yes, you are missing the `files.dll` in Rdsspy. Please copy it from Magic RDS folder.

But due to lack of time I was focusing in recording the file, so I can't tell you if Rdsspy can decode it live or just with wave files.

sm3ulc, 2023-09-17:
Hi!

Maybe I can provide some enlightenment about the development of RDS2 and
the slow takeup reasons. I was involved in RDS-Forum from 2014 to 2022 and
kept track of possibilities for the Swedish Radio. The majority of
receivers are placed in cars. In Sweden a car has an average lifetime of
about 17 years which makes adaptationrate very slow. The national radio in
Sweden are also already using the upper part of some of the channels with
DARC (https://www.windytan.com/2013/11/broadcast-messages-on-darc-side.html).
The standardisation process was also very slow in a landscape where many
stations are contemplating the choice of to go for DAB only/mixed/not at
all.

The _only_ big player I've seen made any effort to actually have done
something with RDS2 is Radio France. They had a working concept running in
2016 and continued  (ex
https://www.radioworld.com/global/radio-france-displays-rds2-file-transfer).

There's also a lack of open reference implementation of the standard that
probably could have helped the process forward.

One basic implementation of the extended carriers:
https://github.com/Anthony96922/MiniRDS

Kind regards,

// David

veso266, 2023-09-17:
@andimik I copied files.dll and curl.exe from Magic RDS to RDS Spy (latest 1.08 version), although RDS Spy can see the plugin when I click on it nothing happens

I also tried to run rds spy which is built in Magic RDS and same thing (maybe I should run it from Magic RDS menu itself or something

RDS Spy has an option called Enable RDS2 tunneling but I am not sure what that does (tunnel the 67khz stream somewhere, but where and how to acsess it)

andimik, 2024-02-20:
@windytan

Oona, it seems that Finnish station
Järviradio uses additional carriers, but it is not sure, if this is RDS2. 

https://www.rundfunkforum.de/viewtopic.php?p=1749186#p1749186

[image]

It's at least in Haapavesi on 89.7 MHz.

I guess you live in or near Helsinki, the station can be found on 103.1 MHz there.

windytan, 2024-02-20:
Thanks for the heads-up. Here the station doesn't seem to have additional subcarriers unfortunately.

Since this is still at an idea phase, and it would be kind of complicated at this point in the development to support RDS2, I've made a brainstorming [discussion thread](#102 ).

andimik, 2024-05-18:
There are other stations, but in Australia.

[image]

[49B1_-_2024-05-18_12-56-44.zip](https://github.com/windytan/redsea/files/15362926/49B1_-_2024-05-18_12-56-44.zip)

windytan, 2024-07-31:
Acceptance criteria for PR (see [discussion](https://github.com/windytan/redsea/discussions/102#discussioncomment-10197997)):
* User can enable RDS2 feature with `--rds2`
* If enabled,
  * redsea looks for additional subcarriers
  * some RDS2 subcarrier metadata (chosen by developer's judgement) is displayed as JSON
  * hex output should be kept RDS Spy compatible
* Developer should design how this JSON info is tied to existing RDS1 JSON groups (if at all), based on the RDS2 documentation
* There's a basic test that runs in CI
* Build shows green

closed 2025-03-14 by windytan