windytan/redsea #34: wrong unicode symbol in the getLCDchar method

flux242, 2016-12-04:
"Sie hoeren: KUNGS FEAT. JAMIE N - DON‖T YOU KNOW"

"¤" - this one seems to be also wrong. I haven't checked any further

windytan, 2016-12-04:
Thanks. Which station is this? Do you happen to have any hex data? I'm seeing correctly decoded apostrophes from at least ANTENNE and RADIO 21.

flux242, 2016-12-04:
the table in the getLCDchar does contain wrong symbols. The symbol before 'a' should be ` but in the table it is ‖

the 5th symbol in the table is ¤ but should be $

shouldn't they?

windytan, 2016-12-04:
The character table is the default charset from page 74 of [the 1998 RDS specification](http://www.interactive-radio-system.com/docs/EN50067_RDS_Standard.pdf). I think this could be fixed by supporting repertoire control codes (switching character sets).

windytan, 2016-12-04:
Or perhaps these characters could just be changed...

flux242, 2016-12-05:
well, seems like radio stations are just using the ascii symbols codes and not the ones defined in the standard.

closed 2016-12-05 by flux242

windytan, 2016-12-05:
idea: A config file with per-station workaround preferences (like `"ascii_strings": true`)

flux242, 2016-12-05:
Maybe you don't need to improve the outdated EBU tables standard at all? Just wanted to add that the symbols could be substituted by the postprocessing. I think I could do such substitutions even using jq -> split("‖") | join("`") for the specific station pi. Sorry for the 'spam' issue

windytan, 2016-12-05:
Not spam at all, very useful discussion :) It seems the standard is applied quite liberally in different parts of the world and by different encoders.