windytan/redsea #36: cannot compile.

sebastianheyn, 2017-01-19:
Hi I'm trying to compile this on a orange pi with debian jesse.

I have liquid installed but make fails

hadow -Wuninitialized -pedantic    -g -O2 -MT redsea-redsea.o -MD -MP -MF .deps/redsea-redsea.Tpo -c -o redsea-redsea.o `test -f 'redsea.cc' || echo './'`redsea.cc
In file included from ../src/subcarrier.h:10:0,
                 from ../src/block_sync.h:10,
                 from redsea.cc:24:
../src/liquid_wrappers.h:55:31: error: expected â before â
   SymSync(liquid_firfilt_type ftype, unsigned k, unsigned m,
                               ^
Makefile:400: recipe for target 'redsea-redsea.o' failed

the configure script can find liquid otherwise it would stop.  Any hints?? Thanks in advance

windytan, 2017-01-19:
Thanks. It seems the terminal didn't print that error message correctly. Could you re-run make using  `export LC_ALL=C; make` in the terminal?

windytan, 2017-05-12:
Did you try this? Was this resolved?

windytan, 2017-08-06:
@sebastianheyn - I assume that this bug is no longer relevant, and I will close it tomorrow.

closed 2017-08-29 by windytan

usagi87, 2017-08-30:
I get the same error when using older version of liquid dsp.

windytan, 2017-08-31:
@usagi87 Thanks. How old a version? Also, does the suggestion I mentioned in my first comment clarify the error message in your case?

usagi87, 2017-09-01:
Hey, I trying to use liquid-dsp v1.2 because v1.3 when used to compile redsea gives an error illegal intruction, but when using it gives a above error.

windytan, 2017-09-02:
@usagi87 Does the suggestion I mentioned in my first comment clarify the error message in your case?

usagi87, 2017-09-02:
In file included from ../src/subcarrier.h:29:0,
                 from ../src/block_sync.h:26,
                 from redsea.cc:23:
../src/liquid_wrappers.h:71:31: error: expected ')' before 'ftype'
   SymSync(liquid_firfilt_type ftype, unsigned k, unsigned m,
                                              ^
Makefile:405: recipe for target 'redsea-redsea.o' failed

windytan, 2017-09-02:
Thanks! I've made an attempt at fixing this now. Would be glad to hear if it works.

usagi87, 2017-09-02:
gives a new error
subcarrier.cc: In constructor 'redsea::Subcarrier::Subcarrier(const redsea::Options&)':
subcarrier.cc:125:14: error: 'LIQUID_FIRFILT_RRC' was not declared in this scope
     symsync_(LIQUID_FIRFILT_RRC, kSamplesPerSymbol, kSymsyncDelay,
                      ^
Makefile:433: recipe for target 'redsea-subcarrier.o' failed
make[2]: *** [redsea-subcarrier.o] Error 1

I found away around the problem with liquid-dsp v1.3 so now redsea work with out crashing.