windytan/redsea #39: Rename project :-)

owenson, 2017-03-30:
Been looking for something like this for a couple of weeks and just stumbled across it.  Maybe call it "super RDS decoder" or similar ? :-)

windytan, 2017-03-30:
Thanks :D I assume this is a disguised compliment? In any case, the name is admittedly not very descriptive.

owenson, 2017-03-30:
Of course, great work just hard to find via Google :-)

andimik, 2017-04-15:
You should at least name it ReDSea

windytan, 2017-04-17:
I tried searching for "RDS decoder" - redsea was on the first page of DuckDuckGo results, second page in Google, and third page in Bing search.

windytan, 2017-05-13:
Searching for "rds decoder linux" brings up redsea as the second result on the first page (US Google). Searching for "rds decoder macos" returns redsea as the first result.

closed 2017-05-13 by windytan