windytan/redsea #41: redsea exits - libliquid.so not found

andimik, 2017-04-12:
Could you tell me why redsea exits? I've even have copied libliquid.so to the readsea directory. libsndfile is installed, too.

> ./rtl-rx.sh -f 90.4M
> redsea: error while loading shared libraries: libliquid.so: cannot open shared object file: No such file or directory
> Found 1 device(s):
>   0:  NOXON, DAB Stick, SN: 0
> 
> Using device 0: Terratec NOXON DAB/DAB+ USB dongle (rev 1)
> Found Fitipower FC0013 tuner
> Tuner gain set to 19.70 dB.
> Bandwidth set to automatic resulted in 0 Hz.
> Tuned to 90058000 Hz.
> Oversampling input by: 8x.
> Oversampling output by: 1x.
> Buffer size: 5.99ms
> Exact sample rate is: 1368000.013046 Hz
> Sampling at 1368000 S/s.
> Output at 171000 Hz.
> Signal caught, exiting!
> 
> User cancel, exiting...
>

andimik, 2017-04-12:
Seems you have to be in the src directory in order to call rtl-rx from the upper directory! 

Could you add this description or change the path?

windytan, 2017-04-12:
Thanks for the report. I can't reproduce this unfortunately. How are you installing liquid-dsp? Which operating system / distro is this?

The shell script can admittedly be confusing, and I'm planning on removing it.

andimik, 2017-04-14:
Now it's working :-) 

Seems I had to restart my PC.

closed 2017-04-14 by andimik

windytan, 2025-10-20:
Hi, did you note that the file is called `libliquid.so.1` and not `libliquid.so`?