windytan/redsea #45: configure: error: Could not find liquid-dsp, but liquid-dsp is installed from MacPorts

spkendall, 2017-05-18:
I can't get past autogen/configure/make. It wants to find liquid-dsp, but can't find it.

```
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for main in -lliquid... no
configure: error: in `/Users/spk/Documents/Source/GitHub/redsea':
configure: error: Could not find liquid-dsp (use --without-liquid to disable)
See `config.log' for more details
spkmbp:redsea spk$
```

From config.log:
```
configure:3591: checking for main in -lliquid
configure:3610: gcc -o conftest -g -O2   conftest.c -lliquid   >&5
ld: library not found for -lliquid
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

liquid-dsp really is installed:
```
spkmbp:redsea spk$ port list installed | grep liquid-dsp
Warning: The 'list' action only shows the currently available version of each port. To see installed versions, use the 'installed' action.
liquid-dsp                     @20170513       science/liquid-dsp
spkmbp:redsea spk$ mdfind liquid.h | egrep "\.h$"
/opt/local/include/liquid/liquid.h
/Users/spk/Documents/Source/GitHub/redsea/src/liquid_wrappers.h
spkmbp:redsea spk$ 
```

spkendall, 2017-05-18:
I finally figured out how to work around the problem, but should I have needed to do this?
```
export CPPFLAGS='-I/opt/local/include'
export LDFLAGS='-L/opt/local/lib'
./autogen.sh && ./configure && make
```

windytan, 2017-05-19:
Thanks, apparently MacPorts does not install in the default include paths and we need to add them separately. Will try and see how that is done in autotools.

windytan, 2017-06-03:
Fixed.

closed 2017-06-03 by windytan