windytan/redsea #48: Hex dumps for TODO 7A 8A 9A 11A 12A

darkbyte-ru, 2017-07-03:
First of all thanks for your software! 

I test in on local fm radio and got some debug TODOs.

So I collect hex dumps in 5 minutes for each stantion with TODOs:

[todo.dump.zip](https://github.com/windytan/redsea/files/1120200/todo.dump.zip)

windytan, 2017-07-03:
Thanks, very useful to have more actual data! I had nothing from Russia before.

I was not able to decode the TODO parts in these, unfortunately:

    $ ./src/redsea -h < ~/Downloads/todo.dump/todo8A.dump | grep 8A | wc -l
           0
    $ ./src/redsea -h < ~/Downloads/todo.dump/unimplemented_ODA_app_31076.dump | grep ODA
           0

Some, if not most, of the messages you're seeing about unimplemented group types are probably caused by bit errors (noise). This seems to be the case because the unimplemented group is only seen once or twice, whereas the actual groups are repeated several times per second.

See the discussion about the robustness of RDS error detection in issue #35. Because we are allowing errors of 1 to 2 bits to be corrected, some errors will inevitably slip through undetected.

andimik, 2017-07-03:
Good evening.

I've checked with RDSSpy (just rename them to *.spy). Yes, indeed, the quality is very, very poor in some cases. Please try to get error-free reception first.

todo8A.dump.spy does not include 8A, but 8B due to bad BER (up to 40%).
[image]

todo8A_12A.dump.spy
Very high BER
[image]

todo9A.dump.spy
[image]
ODA AID 3920 and 3130 are not valid
http://www.nrscstandards.org/RBDS/RDS%20ODA%20AIDs.pdf

darkbyte-ru, 2017-07-12:
Oh, sorry. I just noticed, that SDR, on which I receive RDS, currently work without antenna.
I will attach antenna soon and rescan RDS. If there any "TODO" happens, I will reopen issue.

andimik, 2017-07-12:
Yes, please try again and upload maybe all stations you can receive.

andimik, 2017-08-01:
Any news from Russia?

andimik, 2017-09-01:
Oona, I guess you can close it now. No answer from Russian user.

windytan, 2017-09-03:
I think you're right!

closed 2017-09-03 by windytan