windytan/redsea #50: redsea exiting when libliquid.so not found

davidsrsb, 2017-08-29:
Same problem as issue 41, but a restart does not solve it
Ubuntu 16.04 64 bit
libliquid.so is installed to /usr/local/lib/libliquid.so

I get
redsea: error while loading shared libraries: libliquid.so: cannot open shared object file: No such file or directory

windytan, 2017-08-29:
Does this work?

https://unix.stackexchange.com/questions/67781/use-shared-libraries-in-usr-local-lib

> For the current session you can
>
> `export LD_LIBRARY_PATH=/lib:/usr/lib:/usr/local/lib`
>
> or to make the change permanent you can add `/usr/local/lib` to `/etc/ld.so.conf` (or something it includes) and run `ldconfig` as root.

davidsrsb, 2017-08-30:
/etc/ld.so.conf.d/libc.conf already contained a /usr/local/lib entry, so I ran sudo ldconfig and the library is now found

windytan, 2017-08-30:
Great!

closed 2017-08-30 by windytan