windytan/redsea #52: label mismatch in event_list.h

julpapa31, 2017-10-10:
Hello,

First I would like to say a very very big thank you for your work. Redsea is working like a charm. It's fully support the two commercial TMC services in my local area. 

It's seem that the line number 1062 in event_list.h should be 
 
 "1356;cold,;cold, minimum temperature (Q);F;6;L;2;;37;M6F"

instead of : 

  "1356;cold,;cold, maximum temperature (Q);F;6;L;2;;37;M6F"

About TMC encryption :  in my case they do not encrypted the location, they just send a different table code that the one expected in my country. Changing the table code in my side to the trick.

Best Regards

windytan, 2017-10-10:
Thanks, glad to hear your report about redsea working!

Is this suggestion based on observation of actual transmissions? Note that there is already a message for "cold, minimum temperature", on line 1076. It has a different code (1401, M19F).

julpapa31, 2017-10-10:
It's not based on observation, here the TMC providers don't broadcast any weather information.

Oups, sorry my mistake! You're definetly right, I am translating the event file, and I "cross mistmatch" the event 1401 and 1356.

julpapa31, 2017-10-11:
So I close the issue. Thanks again to share your work.

closed 2017-10-11 by julpapa31