windytan/redsea #53: No RDS test

rajil, 2017-11-12:
I dont get any RDS text. Is there anything i am missing?

```$ rtl_fm -f 95.7e6 -M wbfm -s 200000 -r 48000 - |./redsea -u --feed-through|aplay  -f S16_LE -r48000 
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to automatic.
Tuned to 96016000 Hz.
Oversampling input by: 6x.
Oversampling output by: 1x.
Buffer size: 6.83ms
Sampling at 1200000 S/s.
Output at 200000 Hz.
Playing raw data 'stdin' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
```

andimik, 2017-11-12:
I could not find any station with this PI Code. 

What about other frequencies?

rajil, 2017-11-12:
Here is another one
```
$ rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 40 -F 9 -f 97.9e6 |./redsea   
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 40.20 dB.
Tuned to 98242000 Hz.
Oversampling input by: 8x.
Oversampling output by: 1x.
Buffer size: 5.99ms
Exact sample rate is: 1368000.013046 Hz
Sampling at 1368000 S/s.
Output at 171000 Hz.
{"group":"2A","pi":"0x1511","prog_type":"Varied","tp":false}
{"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x1511","prog_type":"Varied","ta":false,"tp":false}
{"debug":["TODO 8A"],"group":"8A","pi":"0x1511","prog_type":"Varied","tp":false}
{"di":{"dynamic_pty":true},"group":"0A","is_music":true,"pi":"0x1511","prog_type":"Varied","ta":false,"tp":false}
{"group":"3A","open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"pi":"0x1511","prog_type":"Varied","tmc":{"system_info":{"is_encrypted":true,"is_on_alt_freqs":false,"scope":["urban"]}},"tp":false}
{"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x1511","prog_type":"Varied","ta":false,"tp":false}
{"group":"8A","pi":"0x1511","prog_type":"Varied","tp":false}
{"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x1511","prog_type":"Varied","ta":false,"tp":false}
```

and another one
```
$ rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 40 -F 9 -f 104.10e6 |./redsea 
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 40.20 dB.
Tuned to 104442000 Hz.
Oversampling input by: 8x.
Oversampling output by: 1x.
Buffer size: 5.99ms
Exact sample rate is: 1368000.013046 Hz
Sampling at 1368000 S/s.
Output at 171000 Hz.
{"di":{"dynamic_pty":true},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ta":false,"tp":false}
{"group":"2A","pi":"0x3D02","prog_type":"Varied","tp":false}
{"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ta":false,"tp":false}
{"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ta":false,"tp":false}
{"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ta":false,"tp":false}
{"di":{"dynamic_pty":true},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ps":" 104.1  ","ta":false,"tp":false}
{"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ta":false,"tp":false}
{"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ta":false,"tp":false}
{"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ta":false,"tp":false}
{"group":"2A","pi":"0x3D02","prog_type":"Varied","tp":false}
{"di":{"dynamic_pty":true},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ps":" 104.1  ","ta":false,"tp":false}
{"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ta":false,"tp":false}
{"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ta":false,"tp":false}
{"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ta":false,"tp":false}
{"di":{"dynamic_pty":true},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ps":" 104.1  ","ta":false,"tp":false}
{"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ta":false,"tp":false}
{"group":"2A","pi":"0x3D02","prog_type":"Varied","tp":false}
{"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ta":false,"tp":false}
{"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ta":false,"tp":false}
{"di":{"dynamic_pty":true},"group":"0A","is_music":true,"pi":"0x3D02","prog_type":"Varied","ps":" 104.1  ","ta":false,"tp":false}
{"group":"2A","pi":"0x3D02","prog_type":"Varied","tp":false}
```
Is it possible to get RDS with wbfm?

windytan, 2017-11-12:
Hi, I believe the wbfm demodulator removes the RDS subcarrier. This is why the fm demodulator should be used instead. The wbfm demodulator applies a deemphasis filter and may even change the sampling rates.

windytan, 2017-11-13:
Is there a specific reason why you would need the wbfm demodulator? If you want FM stereo or de-emphasis, I've written a simple (yet not very user friendly) tool for that purpose: [stereodemux](https://github.com/windytan/stereodemux).

rajil, 2017-11-13:
I was trying to capture broadcast FM using softFM
(https://github.com/jorisvr/SoftFM), and feed it into icecast. If the
RDS text could be made available that would make the icecast stream more
powerful.

windytan, 2017-12-23:
SoftFM decodes stereo by default. But redsea requires an undecoded mono multiplex. So you would have to pass `-M` or `--mono` to SoftFM to make it work.

I've added a mention in the wiki about this requirement, so closing this for now.

closed 2017-12-23 by windytan