windytan/redsea #54: Decode call sign from PI

classicfm, 2017-11-13:
According to the NRSC-4 standard for the U.S., the PI for a station is based on its call sign, according to a formula in the standard : http://www.nrscstandards.org/SG/nrsc-4-B.pdf (see section 7.12). Most stations here in the U.S. (commercial ones, in any case) use PS either for a slogan (as suggested by the NRSC document) or as a 'poor man's RadioText+'. This leads to interesting results when a radio manufacturer doesn't account for this (the radios in BMWs have had this issue before). 

It would be nice, perhaps when the -u switch is used, to have the PI decoded into a string like 'WFOO' or 'KBAZ'. This is makes things more consistent when trying to use redsea output to identify stations.

windytan, 2017-11-14:
Thanks, great idea!

windytan, 2017-11-14:
There's now preliminary support for this. A JSON field `callsign` will appear in the data when the `-u` option is used. I've tested it to work with the examples provided in that PDF - hope it works with real signals as well!

And by the way, I don't have any real data from the US. I'd be grateful for a minute's worth of hex output of some stations with the `-x` switch, if you have time :)

classicfm, 2017-11-14:
Wow, that was fast! Testing shows me that it works in nearly all cases. One case, where the proper call sign is WYYY, shows up as "KCMQ". It is very possible that the station is simply transmitting incorrect information: this happens more often than you might think (the AF for WCNY is transmitted as 89.7 but is actually 89.5, and they don't put their third alternate in there at all). I have attached some recordings as you requested, including from WYYY. All the others show up correctly. Nice work, and thank you.

By the way, if you are interested, WYYY transmits TMC data, as do most Clear Channel Communications-owned radio stations in the U.S. The data is encrypted, but may be of interest to you since I know redsea does handle TMC data. 

Another possible interesting US-specific bit is emergency alerts. https://en.wikipedia.org/wiki/Emergency_Alert_System These can be delivered by RDS, I believe. Whether they are or not, I don't know.

[WAER-FM.txt](https://github.com/windytan/redsea/files/1472736/WAER-FM.txt)
[WCNY-FM.txt](https://github.com/windytan/redsea/files/1472737/WCNY-FM.txt)
[WNTQ-FM.txt](https://github.com/windytan/redsea/files/1472738/WNTQ-FM.txt)
[WTKW-FM.txt](https://github.com/windytan/redsea/files/1472739/WTKW-FM.txt)
[WYYY-FM.txt](https://github.com/windytan/redsea/files/1472740/WYYY-FM.txt)
[WZUN-FM.txt](https://github.com/windytan/redsea/files/1472741/WZUN-FM.txt)

windytan, 2017-11-15:
WYYY has PI code 1690 which indeed decodes to KCMQ, as if they didn't follow the call sign encoding practice.

Thanks a lot for the data, very useful!

windytan, 2017-11-15:
I just noticed this is actually covered by that document . Stations may use a 1 as the first nibble for better compatibility with a wider range of TMC receivers. They don't offer any easy fix for detecting this. Perhaps the field name could be changed to `callsign_uncertain` or something when the PI starts with 1 and there is a TMC app present.