windytan/redsea #55: Support for other SDR devices

kantooon, 2017-12-22:
Hello,
How hard would it be to have support for other SDR devices like the LimeSDR? Basically I want to create a script that looks simultaneously at all FM channels in a 20 MHz band, decodes the RDS text for each station and feeds it to a webpage (automotive project).

Thanks,
Adrian

windytan, 2017-12-23:
Interesting idea, thanks! For this, redsea would also need a built-in FM demodulator. This wouldn't be hard to do, and it would be generally useful as well. But doing multiple of those in parallel would need some changes made in the way the groups are printed etc.

kantooon, 2017-12-23:
Ok, let me know if you want to make steps in this direction. I can
handle demodulation with GNUradio (pfb_channelizer or the like).

Adrian

sm3ulc, 2017-12-23:
Something you have to build yourself but quite possible. For example:
rxtools => splice flow with ncat => (csdr (for demod) => redsea. ) x channel

// David

windytan, 2017-12-23:
kantooon, then you may be able to do this already, like David suggested. The exact type of SDR device doesn't really matter. You only need to have your script output FM demodulated PCM at 171 kHz sampling rate for every FM carrier you have. Then you would have a redsea instance running at every output. The question is how much computer power it would take to filter and FM demod that many carriers.

kantooon, 2017-12-25:
I'm not sure I understand. How would the audio channels get piped into redsea? I understand the idea of running multiple instances of redsea, although having only one and outputting a single json might be easier.
Thanks,
Adrian

windytan, 2017-12-26:
I think you're right, a single json stream might be best. It seems you can interleave multiple streams in gnuradio. Perhaps the best solution would be to (just brainstorming here):
* have an `--interleave NUM` or `--channels NUM` option in redsea, to read an interleaved PCM stream
* maybe indicate the current channel number for every JSON object when there's more than one channel
* every channel of a multichannel WAV file would be automatically decoded as individual signals
* possibly in the future these could be run in different threads etc.

kantooon, 2017-12-26:
Other possible solutions to pass the stream to redsea: 

- use FIFOs with vector sinks or,
- use sockets with the TCP/UDP sinks in GNU radio
- GNU radio can also do tagged streams, so you could pass around frequency information, signal levels etc. along side audio samples

Display would be relatively straight-forward after that with a webview or QML pages.

windytan, 2018-03-10:
There is now preliminary support for multi-channel signals in the snapshot version (current master branch). It is activated automatically when multi-channel WAV files are used. For raw PCM input you will have to specify the number of channels manually (`--channels`).

kantooon, 2018-03-26:
Awesome. I'll have to try this with a full broadcast spectrum capture.
Thanks!

windytan, 2018-05-22:
Apropos the original topic, [csdr](https://github.com/simonyiszk/csdr) can be used to demodulate streams from other radios on the command line. For example, I can receive RDS with my AirSpy R2 using this command:

    airspy_rx -d -r /dev/stdout -f 87.9 -a 2500000 -v 15 -m 15 -l 14 |\
      csdr convert_i16_f | csdr fir_decimate_cc 10 0.05 HAMMING |\
      csdr fmdemod_quadri_cf | csdr convert_f_i16 | ./src/redsea -r 250000

I added a note about this in the [wiki](https://github.com/windytan/redsea/wiki/Use-cases).

closed 2018-05-22 by windytan