windytan/redsea #57: MACPORTS check breaks cross compile

athoik, 2017-12-24:
Hi @windytan, 

When cross compiling it is not possible to check the for MACPORTS LD and CF.

```
checking for /opt/local/lib... configure: error: cannot check for file existence when cross compiling
```

Currently in order to build `redsea` for ARM or MIPS I have to disable the check.

```
--- a/configure.ac
+++ b/configure.ac
@@ -36,12 +36,7 @@ AS_IF([test "x$with_liquid" != xno],
        [Could not find liquid-dsp (use --without-liquid to disable)])])])

 MACPORTS_LD=
-AC_CHECK_FILE(/opt/local/lib,
-  [AC_SUBST([MACPORTS_LD], ["-L/opt/local/lib"])])
-
 MACPORTS_CF=
-AC_CHECK_FILE(`,
-  [AC_SUBST([MACPORTS_CF], ["-I/opt/local/include"])])

 SNDFILE=
 AS_IF([test "x$with_sndfile" != xno],
```

Could you please add an option to search for MACPORTS (eg --with-macports ) and check only when enabled?

Thanks,
Athanasios

windytan, 2018-01-09:
Hi, you can now use the configure option `--without-macports` to disable this directory check.

closed 2018-01-09 by windytan

athoik, 2018-01-09:
Works great! Thanks!

commit 9836775a (2018-01-09): --without-macports (fixes #57)