windytan/redsea #58: possible to compile this for windows?

veso266, 2018-03-12:
Hi I wonder if its posible to cross-compile this for windows and also wonder if rdssea can read raw MPX stream from 192khz soundcard like RDS Spy can

Thanks for Anwsering and Best Regards

windytan, 2018-03-12:
Unfortunately I don't have any Windows machines available and I can't answer this straight away.

Redsea can read a raw MPX stream from a 192kHz soundcard. There is an example for how to do it on Linux [in the wiki](https://github.com/windytan/redsea/wiki/Use-cases).

veso266, 2018-03-12:
Hi some progress

after installing msys2 and mingw toolchain and development tools (base-devel msys2-devel) I was able to run autogen.sh script now I already builded liquid-dsp per this tut: https://github.com/cjcliffe/CubicSDR/blob/master/external/liquid-dsp/How_to_build_liquid_dsp_for_Windows.txt

now I just need to somehow reference it when executing configure so I am asking how can I specify path to my build dll and a from configure

Thanks for Anwsering and Best Regards

windytan, 2018-03-13:
The error message can be seen on the next to last line; it means that `libsndfile` was not found.

The `--without-liquid` option builds redsea without DSP support. I'm not sure this is what you want. This makes redsea unable to demodulate the RDS subcarrier and only builds hex decoding support. I should perhaps rename it `--disable-dsp`.

veso266, 2018-03-13:
Thanks will try installing libsndfile (there must be one for MSYS2/MinGW) but I still don't know how to specify path to liquid-dsp when running configure (because I already builded that and now have my dll and .lib file to link agains)

windytan, 2018-03-14:
It seems to be libc or glibc: https://unix.stackexchange.com/questions/228886/gcc-ld-cannot-find-lc

veso266, 2018-03-14:
well now I have a problem because glibc and libc are not uvailable on MinGW. There is msvrc but I don't know what needs to be changed for rdsea to use and build agains: https://stackoverflow.com/questions/6394512/standard-c-library-in-mingw

bra1z, 2018-03-28:
I build it using  msys2 32 bits with gcc 7.3.0
Try this:
Add #define M_PI 3.14159265359 in config.h
Add the files:
redsea\src\liquid\liquid.h
redsea\src\liquid\liquid.internal.h
the make fails in the next line:
g++  -g -O2   -o redsea.exe redsea-redsea.o ............  -lc -lliquid  -lsndfile
copy this line, remove -lc , add -liconv and then it links successfully.

windytan, 2018-03-28:
@bra1z - Great! If there's a way to test whether we're building in msys2, this check could perhaps be added to the autoconf files.

veso266, 2018-03-28:
yep it works but you have to go to src dir first

windytan, 2018-03-28:
Try this:
    curl http://5.35.250.101:8000/radio700-mpx.flac | flac --ogg -d -c - | redsea -r 192000

veso266, 2018-03-28:
hmm that odd I tried with your command but nothing interesting happened then tried ./redsea -v and it echoed redsea 0.16-SNAPSHOT by OH2EIQ so it is working

but when trying to read .spy file containing RDS into it: http://downloads.dxing.si/test.spy

it just sits there doing nothing it also does nothing if also did nothing when trying with mpx stream
```
curl http://5.35.250.101:8000/radio700-mpx.flac | flac --ogg -d -c - | redsea -r 192000
```

windytan, 2018-03-28:
redsea reads standard input, unless a WAV file is given with `-f`. So in your first example it's waiting for the hex data via stdin. (This could be changed in the future though, if it's a bit misleading.)  This should work in Linux/macOS: `./redsea -h < test.spy`

Unfortunately I haven't used PowerShell myself. Could it be that you'd have to set the pipe to some kind of raw mode? According to this document, the pipe is not what you'd expect, as PowerShell only pipes 'objects' https://docs.microsoft.com/en-us/powershell/scripting/getting-started/fundamental/understanding-the-windows-powershell-pipeline?view=powershell-6

mnhauke, 2018-03-28:
You can also cross-compile redsea via MinGW.

I created a mingw package for redsea on OBS:
https://build.opensuse.org/package/show/home:mnhauke:mingw:win64:sdr/mingw64-redsea

The example @windytan mentioned does work fine with windows cmd.exe
curl http://5.35.250.101:8000/radio700-mpx.flac | flac --ogg -d -c - | redsea -r 192000

Here's an archive with precompiled windows  binaries for redsea v0.16.0, flac and curl:
https://pmtu.de/.dump/mingw64-redsea-v0-16-0.zip

veso266, 2018-03-29:
yey works like a charm
@mnhauke Thanks for compiling it so others without build-envs setups can use it

using this command to get only PS
```
curl -s http://5.35.250.101:8000/radio700-mpx.flac | flac --ogg -d -c - | redsea -r 192000 | jq ".ps" 
```(cmd doesn't like ' so " it is)

now I just need to filter nulls out

veso266, 2018-03-29:
I Think it crashes because it reaches end of file
but the error is as follows in libsndfile
```
Ime težave:	APPCRASH
  Program Name:	redsea.exe
  Program Version:	0.0.0.0
  Timestamp:	5abbc1d0
  Library name :	libsndfile-1.dll
  Library version:	1.0.28.0
  Error TimeStamp:	00000000
  Šifra izjeme:	c0000005
  Odmik izjeme:	0000000000001c65
  Različica operacijskega sistema:	6.1.7601.2.1.0.256.1
  Področne nastavitve:	1060
  More Info 1:	bc3a
  More Info 2:	bc3abdd9effc386365dab36d02b1fef6
  More Info 3:	0de5
  More Info 4:	0de50e418839be0272e13b196ca46db0
```

as for TMC is is possible to get RAW data to trow it into some other TMC Decoder like: https://github.com/razor/rds-tmc-decoder or https://github.com/mueschel/TmcDecoder?

veso266, 2018-03-30:
great
How did you get location tables for your TMC provider?
Because I tried to do this with TrafficNav (TMC provider that we are using in Slovenija and Hungary) and they wouldn't send me their LC
they said that they wanted to sign a contract with me and I don't want to sign anything

windytan, 2018-03-30:
This depends entirely on the individual organisation. It's unfortunate that they don't want to give that out, but it could be part of their business model. I got mine a very long time ago by just asking, but it could be that they have changed their policies, too.

You can only use a location table that has the same country code and location table number as in the transmission.

veso266, 2018-03-30:
which TMC provider did you have (probably not TrafficNav)

now the only option would be to extraxt them along with the keys from Navigation system that payed the licenses: http://www.navitotal.com/general-discussions-about-tomtom-f81/rsd-tmc-traffic-decryption-keys-t18115.html

http://download.tomtom.com/sweet/navcore/8.351.go530-go730-go930.CAB-navcore-GO.cab the file that should hold them
now I hope they are in standard format there

bra1z, 2018-03-31:
I found a project with a few  location tables:
https://gitlab.com/mvglasow/qz
Look in asset folder. The files are in jdbc format.
By the way, I found a really easy way to view the TMC locations using the node-red program.
I use the node daemon to run the redsea program and the node worldmap to see the points.

veso266, 2018-03-31:
nice
but I am not sure if theese LC are private or public ones 
can you share your script to plot on a Map with node and Redsea?

bra1z, 2018-03-31:
Yeah, sure
I use the next cmd script:
`C:\rtl_fm\rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 10 -F 9 -f 93.2M | C:\rtl_fm\redsea -l TMC
`
Check the daemon node properties for write the correct path to the script.
Note I am using absolute paths ( maybe this is not necessary using linux or Mac ).
 If you have problems installing red-node, the electron version is more easy to install:
https://electronjs.org/apps/node-red
Remember, you have also to install  node-red-node-daemon and node-red-contrib-web-worldmap.
Don't forget type Ctrl. Shift M to see the map. This is a very simple flow, doesn't draw all points.
Copy this flow JSON to your clipboard and then import into Node-RED using the Import From > Clipboard (Ctrl-I) menu option 

`[{"id":"bb9c107d.61df","type":"daemon","z":"dc7c50ab.e5301","command":"C:\\rtl_fm\\run.cmd","args":"","cr":false,"redo":true,"op":"string","name":"REDSEA","x":100,"y":80,"wires":[["a1f033a1.8a46f"],[],[]]},{"id":"cde22226.0c2f7","type":"function","z":"dc7c50ab.e5301","name":"extract plots","func":"var value = msg.payload.group;\n\nif(value==\"8A\")\n{\n    if(\"tmc\" in msg.payload)\n    {\n    var tmc_object = msg.payload.tmc;\n\n        if(\"message\" in tmc_object)\n        {\n        var message_object = tmc_object.message;\n        \n            msg.payload = { \n                    name : message_object.road_name + \" \" + message_object.location,\n            \n                    lat  : parseFloat(message_object.coordinates[0].lat),\n                    lon  : parseFloat(message_object.coordinates[0].lon),\n            };\n        \n         return msg;\n        }\n    }\n}\n","outputs":1,"noerr":0,"x":210,"y":260,"wires":[["f78b623d.80716"]]},{"id":"a1f033a1.8a46f","type":"json","z":"dc7c50ab.e5301","name":"","pretty":false,"x":230,"y":180,"wires":[["cde22226.0c2f7"]]},{"id":"f78b623d.80716","type":"worldmap","z":"dc7c50ab.e5301","name":"","lat":"","lon":"","zoom":"","layer":"","cluster":"","maxage":"","usermenu":"show","panit":"false","x":190,"y":380,"wires":[]}]`

bra1z, 2018-05-09:
I made a program for Windows in C# language using redsea. The program draw the plots of RDS-TMC traffic.  Be sure to put in TMC folder the correct location tables and edit the file run.cmd to change frequency and gain. https://github.com/bra1z/redseaTMCmap

windytan, 2018-06-05:
The original issue being resolved by the mingw build I'm closing this issue now. I added a mention about this in [the wiki](https://github.com/windytan/redsea/wiki/Installation).

closed 2018-06-05 by windytan

master131, 2022-02-19:
Just in-case anyone looks at this issue again, I've added additional ways which redsea can be installed on Windows on the Wiki:
https://github.com/windytan/redsea/wiki/Installation

These include:
- Cross-compiling using MinGW as already previously mentioned
- Using the Windows Subsystem for Linux (warning: there is currently no support for libusb in WSL2)
- Using Cygwin

@windytan FYI.