windytan/redsea #59: No output from redsea

nailgg, 2018-04-15:
I can't get any output from redsea. Tried on x64 Ubuntu and Raspberry Pi 3, I couldn't get redsea to work on both devices.

```
rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 40 -F 9 -f 100.0e6 - | redsea
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 40.20 dB.
Tuned to 98242000 Hz.
Oversampling input by: 8x.
Oversampling output by: 1x.
Buffer size: 5.99ms
Exact sample rate is: 1368000.013046 Hz
Sampling at 1368000 S/s.
Output at 171000 Hz.
```
Is all I get.

Tried `-M wbfm` and `-M fm` flags on `rtl_fm`, both didn't work.
Tried changing the frequency, bandwidth, gain values etc., no matter what I tried, I couldn't get anything.
Tried the `-x` flag on redsea but it didn't spit anything.

On Windows, I am able to view the RDS data using `SDR# + RDS Data Logger` so I am sure there RDS data is available around me. What am I doing wrong?

nailgg, 2018-04-16:
Finally, running it through the line below worked:

`{ rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 40 -F 9 -f 106.2M | redsea -e 2>&1 1>&3 3>&- | jq -c . ; } 3>&1 1>&2 | aplay -r 171k -f S16_LE`

closed 2018-04-16 by nailgg

project-owner, 2018-10-02:
pi@raspberrypi:~ $ redsea --version
redsea 0.17.1 by OH2EIQ

Here is the output using that syntax (no RDS messages):
pi@raspberrypi:~ $ { rtl_fm -M fm -l 0 -A std -s 171k -g 20 -F 9 -f 106.9M | redsea -e 2>&1 1>&3 3>&- | jq -c . ; } 3>&1 1>&2 | aplay -r 171000 -f S16_LE
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Tuner gain set to 19.70 dB.
Tuned to 107242000 Hz.
oversampling input by: 8x.
Oversampling output by: 1x.
Buffer size: 5.99ms
Exact sample rate is: 1368000.013046 Hz
Create UDP thread
Created UDP thread
Main socket started! :-) Tuning enabled on UDP/6020 
Sampling at 1368000 S/s.
Output at 171000 Hz.
Playing raw data 'stdin' : Signed 16 bit Little Endian, Rate 171000 Hz, Mono