windytan/redsea #60: Decoding not working after release v0.15

sm3ulc, 2018-06-07:
Up to v0.15 it's working like charm but after the we are not able to get any output whatsoever. OS is Ubuntu (18) but same result on OSX. 

redsea-0.15.0/src# rtl_fm -M fm -l 0 -A std -p -52 -s 171k -g 0 -F 9 -f 103.3M | ./redsea -E
Found 1 device(s):
  0:  NOXON, DAB Stick, SN: 0
Using device 0: Terratec NOXON DAB/DAB+ USB dongle (rev 1)
Found Fitipower FC0013 tuner
Tuner gain set to -5.40 dB.
Tuner error set to -52 ppm.
Tuned to 103642000 Hz.
Oversampling input by: 8x.
Oversampling output by: 1x.
Buffer size: 5.99ms
Exact sample rate is: 1368000.013046 Hz
Sampling at 1368000 S/s.
Output at 171000 Hz.
{"bler":2,"debug":["TODO 10A"],"group":"10A","pi":"0xEC24","prog_type":"Information","tp":true}
....

But on 0.16 an onwards we get no data. We tried also with '-c 0/1/2' in case it maybe was related with the support for multichannel.

redsea-0.16.0/src# rtl_fm -M fm -l 0 -A std -p -52 -s 171k -g 0 -F 9 -f 103.3M | ./redsea -E
Found 1 device(s):
  0:  NOXON, DAB Stick, SN: 0
Using device 0: Terratec NOXON DAB/DAB+ USB dongle (rev 1)
Found Fitipower FC0013 tuner
Tuner gain set to -5.40 dB.
Tuner error set to -52 ppm.
Tuned to 103642000 Hz.
Oversampling input by: 8x.
Oversampling output by: 1x.
Buffer size: 5.99ms
Exact sample rate is: 1368000.013046 Hz
Sampling at 1368000 S/s.
Output at 171000 Hz.

Any ideas of how to troubleshoot further?

windytan, 2018-06-08:
Thanks for the report. Could you possibly upload 10 seconds worth of output from that rtl_fm command somewhere for testing purposes?

sm3ulc, 2018-06-08:
Trying to do an attachement.

// David

sm3ulc, 2018-06-08:
Pulled latest master. Works like a charm again. :) Thanks for great support and software! Using redsea as one of the tools to verify a fresh implemenation of a RDS-encoder.

windytan, 2018-06-08:
Great to hear it's being used in real-life applications!

closed 2018-06-08 by windytan