windytan/redsea #62: no output

project-owner, 2018-09-30:
No output from redsea, I've tried about dozen different stations here in San Francisco Bay Area but no avail.
Here is the syntax which I tried:
rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 40 -F 9 -f 96.5M | redsea -e | aplay -r 171000 -f S16_LE
rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 40 -F 9 -f 96.5M | redsea -e -u | aplay -r 171000 -f S16_LE
I can hear audio but no output from redsea in terminal. I also tried weird syntax suggested here:
https://github.com/windytan/redsea/issues/59
but still no output. Just wondering if I'm doing something wrong. 
Thanks in advance!

windytan, 2018-10-01:
Thanks for the report!

It doesn't look like you're doing anything wrong. A couple of questions:
- Are you able to get output from redsea without the -e option (and the pipe to aplay)?
- Is this a Linux system?

project-owner, 2018-10-01:
No output without aplay either:
rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 40 -F 9 -f 96.5M | redsea
This is Raspbian Stretch Desktop running on Raspberry Pi 3+. I'm using the latest version of redsea.
I tried the same stations/frequencies on Windows machine with SDR# and I saw the RDS messages there
for the same stations.
Thanks!

project-owner, 2018-10-01:
Here is the output:
pi@raspberrypi:~ $ rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 40 -F 9 -f 97.3M | redsea
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Tuner gain set to 40.20 dB.
Tuned to 97642000 Hz.
oversampling input by: 8x.
Oversampling output by: 1x.
Buffer size: 5.99ms
Exact sample rate is: 1368000.013046 Hz
Create UDP thread
Created UDP thread
Main socket started! :-) Tuning enabled on UDP/6020 
Sampling at 1368000 S/s.
Output at 171000 Hz.

windytan, 2018-10-02:
You could try to change the gain (-g) to -g 20.

If that doesn't help, then I'd be interested in seeing a short (10-second) recording of that raw rtl_fm output.

sm3ulc, 2018-10-02:
Hi!

You have set ppm to 0. What dongle do you have? Try to calibrate/measure
your ppm if not done earlier.

// David

project-owner, 2018-10-02:
I tried two different dongles with the same result:
https://www.amazon.com/NooElec-NESDR-SMArt-Bundle-R820T2-Based/dp/B01GDN1T4S
https://www.amazon.com/RTL-SDR-Blog-RTL2832U-Software-Telescopic/dp/B011HVUEME
I'm not sure how to calibrate the dongle. If I remove that parameter the result is the same.

I recorded three test files for different stations this way:
rtl_fm -M fm -l 0 -A std -s 171k -g 20 -F 9 -f 106.9M > test-106.9M.raw
They can be found here on github:
https://github.com/project-owner/Peppy.doc/blob/master/files/test-102.9M.raw
https://github.com/project-owner/Peppy.doc/blob/master/files/test-106.9M.raw
https://github.com/project-owner/Peppy.doc/blob/master/files/test-97.3M.raw

Please let me know if it works or if I need to use any other way to record those files.
Thanks!

windytan, 2018-10-02:
Thanks for the files, they work well. I was able to decode them in redsea v0.16 and v0.17.1 on MacOS (I got callsigns KCBS, KLLC, KBLX). To me this looks very similar to the v0.15 bug you linked to earlier. Which version does `redsea --version` report?

project-owner, 2018-10-02:
pi@raspberrypi:~ $ redsea --version
redsea 0.17.1 by OH2EIQ

Here is the output using that syntax (no RDS messages):
pi@raspberrypi:~ $ { rtl_fm -M fm -l 0 -A std -s 171k -g 20 -F 9 -f 106.9M | redsea -e 2>&1 1>&3 3>&- | jq -c . ; } 3>&1 1>&2 | aplay -r 171000 -f S16_LE
Found 1 device(s):
0: Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Tuner gain set to 19.70 dB.
Tuned to 107242000 Hz.
oversampling input by: 8x.
Oversampling output by: 1x.
Buffer size: 5.99ms
Exact sample rate is: 1368000.013046 Hz
Create UDP thread
Created UDP thread
Main socket started! :-) Tuning enabled on UDP/6020
Sampling at 1368000 S/s.
Output at 171000 Hz.
Playing raw data 'stdin' : Signed 16 bit Little Endian, Rate 171000 Hz, Mono

andimik, 2019-08-12:
@project-owner 

Is this solved now?

project-owner, 2019-08-12:
I didn't try after that report. Is there any new redsea release which could fix it? I could lend Pi3 + RTL-SDR dongle to anybody who could fix the issue.

windytan, 2019-08-12:
There has been a release after this, but it is difficult to say whether it would fix this problem, as the root cause is not known. Perhaps!

project-owner, 2019-08-13:
Updated OS, recompiled rtl-sdr and installed new version of redsea: redsea 0.18 by OH2EIQ
It works now. I can see the RDS Json output. Now I need to find a way to capture that output in my program (maybe using fifo/named pipe ?) Also need a way to redirect audio output to VLC which I control from my program. But that's a different story. Closing the issue.

closed 2019-08-13 by project-owner