windytan/redsea #63: multiple TMC location tables

veso266, 2018-10-10:
Hi I was wondering if it is possible to load TMC location tables from multiple countries into redsea

I can recieve 2 public TMC services on from Slovenia and one from Italy and both carry TMC, but the one from italy of course would need italien location tables to show location corecly

I am not sure how GPS vendors do that
one scenario was when I was in Croatia and recieve RAI 1 from Italy and the car would load Italien location tables and show location from Italy that was like 200km away :smile:

windytan, 2018-10-11:
Currently this is possible if you stop and re-start redsea with different command-line parameters. You would need to do this anyway to change the reception frequency, depending on how exactly you're using redsea.

Each location table has a country code and location table number, and a receiver should be able to tell whether it has the correct table for a TMC service or not.

veso266, 2018-10-11:
I am using redsea in a C# app feeding it RDS data with HEX so when tuning SDR# rds data changes so I do not need to restart redsea, not sure how would I feed multiple location tables in it?

windytan, 2018-10-11:
In that case it's not possible as of yet, but sounds like it could very well be implemented.

commit e265c683 (2018-11-11): add support for loading multiple location databases (fixes #63)
closed 2018-11-11 by windytan