windytan/redsea #64: Redsea stops receiving data after two seconds

MCelliotG, 2019-01-13:
Hello everyone, I use rtl_fm and redsea to decode RDS on an Enigma2 receiver (linux based). I use an Astrometa DVB-T2 dongle for receiving radio. I have no problem with general reception and everything works well except RDS. It stops receiving and updating data after 1-2 seconds and if it does it displays only an erroreous PI after finding the correct one.
Noted, that the dongle works great on PC with the same antenna and configuration and SDR programs.
Here's the output from a command line that should work perfect but does not
`root@hd51:~# rtl_fm -f 90.80M -M fm -l 0 -A std -s 171k -g 8 -p -99 -F 9 - | redsea -e
Found 1 device(s):
  0:  astrometadvbt2, dvbt2, SN:

Using device 0: Astrometa DVB-T/DVB-T2
Found Rafael Micro R828D tuner
Tuner gain set to 7.70 dB.
Tuner error set to -99 ppm.
Tuned to 91142000 Hz.
Oversampling input by: 8x.
Oversampling output by: 1x.
Buffer size: 5.99ms
Exact sample rate is: 1368000.013046 Hz
Sampling at 1368000 S/s.
Output at 171000 Hz.
{"pi":"0x10C7"}
{"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0x10C7"}
{"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"dynamic_pty":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"pi":"0x10C7"}
{"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0x10C7"}
{"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"dynamic_pty":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0x12C6"}
{"pi":"0x12C6"}
{"pi":"0xE08C"}
{"pi":"0xE08C"}
{"pi":"0xE08C"}
{"pi":"0xE70E"}
{"pi":"0xE70E"}
{"pi":"0xE70E","prog_type":"National music","tp":false}
{"pi":"0xE70E"}
{"pi":"0xE70E"}

`
The gain is optimal for the reception I have since I use an external antenna. As you can see the initial paramaters are all found except ps but then fallback to erroreous values and decoding stops.

Now if I use the -p parameter
`root@hd51:~# rtl_fm -f 90.80M -M fm -l 0 -A std -s 171k -g 8 -p -99 -F 9 - | redsea -e -p
Found 1 device(s):
  0:  astrometadvbt2, dvbt2, SN:

Using device 0: Astrometa DVB-T/DVB-T2
Found Rafael Micro R828D tuner
Tuner gain set to 7.70 dB.
Tuner error set to -99 ppm.
Tuned to 91142000 Hz.
Oversampling input by: 8x.
Oversampling output by: 1x.
Buffer size: 5.99ms
Exact sample rate is: 1368000.013046 Hz
Sampling at 1368000 S/s.
Output at 171000 Hz.
{"di":{"artificial_head":false},"group":"0A","is_music":true,"partial_ps":"  N     ","pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"compressed":false},"group":"0A","is_music":true,"partial_ps":"  N AI  ","pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"dynamic_pty":false},"group":"0A","is_music":true,"partial_ps":"  N AIR:","pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"group":"2A","partial_radiotext":"                MUSI","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"group":"2A","partial_radiotext":"                MUSIC ON","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"di":{"stereo":false},"group":"0A","is_music":true,"partial_ps":" O      ","pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"artificial_head":false},"group":"0A","is_music":true,"partial_ps":" ON     ","pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"compressed":false},"group":"0A","is_music":true,"partial_ps":" ON AI  ","pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"dynamic_pty":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ps":" ON AIR:","ta":false,"tp":true}
{"group":"2A","partial_radiotext":"                MUSIC ONLY","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"group":"2A","partial_radiotext":"                MUSIC ONLY","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"di":{"stereo":false},"group":"0A","is_music":true,"partial_ps":"LA      ","pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"artificial_head":false},"group":"0A","is_music":true,"partial_ps":"LADY    ","pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"di":{"dynamic_pty":false},"group":"0A","is_music":true,"partial_ps":"      AG","pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0x10C7"}
{"pi":"0x10C7"}
{"pi":"0x10C7"}
`
In this case the ps is found partially and the PI does not fallback to erroreous values, but the decoding stops again.
I don't know what to do and if I do something wrong, it happens with all RDS stations and especially if they have an insensitive pilot for RDS nothing is received because it takes up more time and redsea stops after 2 seconds.
What could be the issue?

windytan, 2019-01-15:
Thanks for the report. Repeated erroneous PI codes sounds pretty bad. Is there any possibility you could upload 10 seconds worth of the raw MPX signal somewhere, for further investigation?

MCelliotG, 2019-01-15:
Thank you for your reply! Could you provide me a recommended command to do this?

windytan, 2019-01-16:
For example:

    rtl_fm -f 90.80M -M fm -l 0 -A std -s 171k -g 8 -p -99 -F 9 - > mpx.raw

Let it run for 10 seconds and then interrupt it with Ctrl+C. The resulting file is called `mpx.raw`.

Also, it would be interesting to know whether the release version works for you (v0.17.1). It can be downloaded here: https://github.com/windytan/redsea/releases/tag/v0.17.1

MCelliotG, 2019-01-16:
Thank you, here's the raw file you asked.
https://mega.nz/#!mJ1lhSSY!JnR1iFMK8Y50loVdbZCJ_fZD-kN_pfY79-YgfM4Zd8g

I cannot test version 17 ftm, but I have also version 16 and it works quite well (it's a bit slow with radiotext messages but is working).

windytan, 2019-01-19:
Great, thanks. I've tested the file under Ubuntu Linux (Intel i7), macOS (Intel i7), raspbian (armv7) and it seems to work all right, outputting 646 lines of JSON and all the PI codes are 0x10C7. So this is still a mystery.

I wonder if your Enigma2 system has a PowerPC CPU? What does the command `uname -a` output?

MCelliotG, 2019-01-19:
satdreamgr 6 hd51

hd51 login: root
That's good to know, I wonder about that too, version 16 (these are the only ones I can test since I cannot compile redsea myself for these systems) works quite well, but it's a bit slower at detecting data.  The problem is with version 18.
Now my system has these specs:
root@hd51:~# uname -a
Linux hd51 4.10.12 #1 SMP Wed Sep 26 22:22:36 EEST 2018 armv7l GNU/Linux
root@hd51:~#

MCelliotG, 2019-01-22:
I believe I found the root to the problem. 
I use two different Enigma2 images, SatDreamGr (Openpli) and OpenATV.
SatDreamGr has libsndfile1 version 1.0.27-r0, OpenATV has version 1.0.28.
On OpenATV RDS works perfectly, receives data constantly with no loss and PI changes, regardless the redsea version.
On SatDreamGr the data stops after 2 seconds regardless the redsea version. So, obviously libsndfile plays a crucial part with RDS reception!

andimik, 2019-01-22:
Congratulations, that you found that out.

I always thought that the Greek image bases on OpenATV...

MCelliotG, 2019-01-22:
Nope, SDG was always based on OpenPLI, I discovered that by accident, since I initially thought the redsea version was the problem. So I took the redsea 0.18 binary and placed it on OpenATV image replacing version 0.16. Then the console nagged that the libsndfile libraries were missing. I installed them from their feeds (1.0.28) and boom!!! Perfect RDS! 
What suprises me though is that on OpenATV redsea version 0.16 was working without (updated) libsndfile libraries. Anyway, if there's not anything else that affects RDS, I think SDG only need to update libsndfile1 and it should be fine on their image too!

andimik, 2020-02-01:
I guess you can close this issue

MCelliotG, 2020-02-01:
Yes, of course!

closed 2020-02-01 by MCelliotG