windytan/redsea #65: Cannot compile on ARMv7

takgr, 2019-04-23:
Hi Oona and thanks for this beautiful software!
Trying to compile on ARMv7 (Allwiner H2+) fails with the following error:

make  all-recursive
make[1]: Entering directory '/root/redsea'
Making all in src
make[2]: Entering directory '/root/redsea/src'
g++ -DHAVE_CONFIG_H -I. -I..  -std=c++14 -Wall -Wextra -Wstrict-overflow -Wshadow  -Wuninitialized -pedantic    -g -O2 -MT redsea-block_sync.o -MD -MP -MF .deps/redsea-block_sync.Tpo -c -o redsea-block_sync.o `test -f 'block_sync.cc' || echo './'`block_sync.cc
In file included from ../src/groups.h:36:0,
                 from ../src/block_sync.h:24,
                 from block_sync.cc:17:
../src/tmc/tmc.h:152:40: warning: missing initializer for member ‘std::array<short unsigned int, 2u>::_M_elems’ [-Wmissing-field-initializers]
   std::array<uint16_t, 2> data        {};
                                        ^
../src/tmc/tmc.h:195:38: warning: missing initializer for member ‘std::array<redsea::tmc::MessagePart, 5u>::_M_elems’ [-Wmissing-field-initializers]
   std::array<MessagePart, 5> parts_ {};
                                      ^
In file included from ../src/block_sync.h:24:0,
                 from block_sync.cc:17:
../src/groups.h:155:33: warning: missing initializer for member ‘std::array<redsea::Block, 4u>::_M_elems’ [-Wmissing-field-initializers]
   std::array<Block, 4> blocks_ {};
                                 ^
block_sync.cc: In function ‘constexpr redsea::eBlockNumber redsea::BlockNumberForOffset(redsea::Offset)’:
block_sync.cc:31:19: warning: compound-statement in constexpr function [-Wpedantic]
   switch (offset) {
                   ^
block_sync.cc:42:1: error: body of constexpr function ‘constexpr redsea::eBlockNumber redsea::BlockNumberForOffset(redsea::Offset)’ not a return-statement
 }
 ^
block_sync.cc: In function ‘constexpr redsea::Offset redsea::NextOffsetFor(redsea::Offset)’:
block_sync.cc:46:24: warning: compound-statement in constexpr function [-Wpedantic]
   switch (this_offset) {
                        ^
block_sync.cc:56:1: error: body of constexpr function ‘constexpr redsea::Offset redsea::NextOffsetFor(redsea::Offset)’ not a return-statement
 }
 ^
block_sync.cc: In function ‘constexpr redsea::Offset redsea::OffsetForSyndrome(uint16_t)’:
block_sync.cc:60:21: warning: compound-statement in constexpr function [-Wpedantic]
   switch (syndrome) {
                     ^
block_sync.cc:69:1: error: body of constexpr function ‘constexpr redsea::Offset redsea::OffsetForSyndrome(uint16_t)’ not a return-statement
 }
 ^
block_sync.cc: In member function ‘void redsea::SyncPulseBuffer::Push(redsea::Offset, int)’:
block_sync.cc:175:17: error: no match for ‘operator=’ (operand types are ‘std::array<redsea::SyncPulse, 4u>::value_type {aka redsea::SyncPulse}’ and ‘<brace-enclosed initializer list>’)
   pulses.back() = {offset, bitcount};
                 ^
block_sync.cc:175:17: note: candidates are:
In file included from block_sync.cc:17:0:
../src/block_sync.h:29:8: note: redsea::SyncPulse& redsea::SyncPulse::operator=(const redsea::SyncPulse&)
 struct SyncPulse {
        ^
../src/block_sync.h:29:8: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘const redsea::SyncPulse&’
../src/block_sync.h:29:8: note: redsea::SyncPulse& redsea::SyncPulse::operator=(redsea::SyncPulse&&)
../src/block_sync.h:29:8: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘redsea::SyncPulse&&’
In file included from ../src/tmc/tmc.h:36:0,
                 from ../src/groups.h:36,
                 from ../src/block_sync.h:24,
                 from block_sync.cc:17:
../src/util.h: In instantiation of ‘redsea::RunningSum<N>::RunningSum() [with unsigned int N = 50u]’:
block_sync.cc:195:19:   required from here
../src/util.h:107:16: warning: missing initializer for member ‘std::array<int, 50u>::_M_elems’ [-Wmissing-field-initializers]
   RunningSum() {};
                ^
../src/util.h: In instantiation of ‘redsea::RunningAverage<N>::RunningAverage() [with unsigned int N = 12u]’:
block_sync.cc:195:19:   required from here
../src/util.h:127:20: warning: missing initializer for member ‘std::array<int, 12u>::_M_elems’ [-Wmissing-field-initializers]
   RunningAverage() {};
                    ^
Makefile:450: recipe for target 'redsea-block_sync.o' failed
make[2]: *** [redsea-block_sync.o] Error 1
make[2]: Leaving directory '/root/redsea/src'
Makefile:410: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/root/redsea'
Makefile:329: recipe for target 'all' failed
make: *** [all] Error 2

 cat /proc/cpuinfo
Processor       : ARMv7 Processor rev 5 (v7l)
processor       : 0
BogoMIPS        : 2400.00

processor       : 1
BogoMIPS        : 2400.00

processor       : 2
BogoMIPS        : 2400.00

processor       : 3
BogoMIPS        : 2400.00

Features        : swp half thumb fastmult vfp edsp thumbee neon vfpv3 tls vfpv4 idiva idivt
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : sun8i
Revision        : 0000

cat /etc/issue
Debian GNU/Linux 8 \n \l

Linux orangepizero 3.4.113-sun8i #8 SMP PREEMPT Sat Feb 9 20:17:57 CET 2019 armv7l GNU/Linux

However on a different board with ARM64 it compiles!

Disabling TMC does not help, am I doing something wrong?

windytan, 2019-04-24:
Thanks for the report. Just to be sure, which compiler is this? (`g++ --version`)

takgr, 2019-04-24:
g++ (Debian 4.9.2-10+deb8u2) 4.9.2

windytan, 2019-04-24:
This is due to incomplete C++14 support in GCC 4.9.

But please have a try; I have now modified these parts of the code to work with 4.9. This change is now in the current master branch.

takgr, 2019-04-25:
It works now!!!!
Thank a lot for such a fast response and fix!
I'm now REaDy to sail the SEA of RDS!!!

closed 2019-04-25 by takgr