windytan/redsea #66: can not compile for macOS 10.14.5

Mike4U, 2019-06-01:
trying to compile release 0.18 using default MacPorts installation.  Get these results:

`xxxx-iMac:redsea-0.18 xxxx$ ./configure
checking for a BSD-compatible install... /opt/local/bin/ginstall -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /opt/local/bin/gmkdir -p
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for symtrack_cccf_create in -lliquid... no
configure: error: in `/Users/michaelrunyan/redsea-0.18':
configure: error: Could not find liquid-dsp (use --without-liquid to disable)
See `config.log' for more details`

However liquid-dsp is installed
`xxxx-iMac:redsea-0.18 xxxx$ port installed liq*
The following ports are currently installed:
  liquid-dsp @20190521-c200ecbd_0 (active)
`

if try without liquid-dsp then:
`xxxx-iMac:redsea-0.18 xxxx$ ./configure --without-liquid
checking for a BSD-compatible install... /opt/local/bin/ginstall -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /opt/local/bin/gmkdir -p
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for /opt/local/lib... yes
checking for /opt/local/lib... (cached) yes
checking for main in -lsndfile... no
configure: error: in `/Users/michaelrunyan/redsea-0.18':
configure: error: Could not find libsndfile
See `config.log' for more details
`

libsndfile is also installed so it must be the path?
`xxxx-iMac:redsea-0.18 xxxx$ echo $PATH
/opt/local/lib:/opt/local/bin:/opt/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin`

Any help would be appreciated.  thanks

windytan, 2019-06-01:
Hi, can you check the values of `MACPORTS_CF` and `MACPORTS_LD` in `config.log`?

Did you run this: `xcode-select --install`

Mike4U, 2019-06-01:
Hi, got it going, just used brew install for liquid-dsp and libsndfile per the wiki.  Don't know why it wouldn't work with the port install.  I had run xcode-select --install

from ~/redsea-0.18/config.log:
MACPORTS_CF='-I/opt/local/include'
MACPORTS_LD='-L/opt/local/lib'

thanks for the help

closed 2019-06-01 by Mike4U

windytan, 2019-06-01:
I'm glad you got it working. Too bad I'm not using MacPorts myself, couldn't easily debug this further. It perhaps seems like something others might come across in the future.

Mike4U, 2019-06-01:
When I figure it out, I'll post something.

thanks again