windytan/redsea #72: RDSSpy recording with new ODA available: iTunes tagging (hex C3B0)

andimik, 2020-08-20:
See https://raw.githubusercontent.com/walczakp/rds-spy-logs/3a45d7c9f26f96c8149fd47dbad067477acd6260/USA/4569%20-%202020-08-19%2020-45-06.spy

This is a PR from https://github.com/walczakp/rds-spy-logs/pull/2

andimik, 2020-08-20:
I found two files from Sweden (recorded through Airspy Server last year) with "Warning receiver" thanks to `egrep -rni -e "ODA" *.txt`

[E424_-_2019-04-29_20-25-12.txt](https://github.com/windytan/redsea/files/5105226/E424_-_2019-04-29_20-25-12.txt)
[E424_-_2019-04-29_20-23-20.zip](https://github.com/windytan/redsea/files/5105230/E424_-_2019-04-29_20-23-20.zip)

```
ODA 1:   AID        = 6A7A
         Proportion = 0,2 %  (1/492)
         App. Group = 12A
```

[E224_-_2019-04-29_19-59-58.txt](https://github.com/windytan/redsea/files/5105240/E224_-_2019-04-29_19-59-58.txt)
[E224_-_2019-04-29_19-58-08.zip](https://github.com/windytan/redsea/files/5105242/E224_-_2019-04-29_19-58-08.zip)

andimik, 2020-08-20:
Assuming that the ODA code is the last block in each line of rdsspy recordings, I used
`egrep -rni -e "CB73|4C59|CC21|1DC2|4AA1|0093|1BDA|0F87|0E2C|1D47|4BA2|E5D7|0C24|CD9E|4AB7|1CB1|4D9A|E319|0E31|CB97|E440|4D95|1E8F|0D8B|E4A6|1C5E|0BCB|CE6B|1DC5|4D87|0D45|CD47|0CC1|1C68|7373|5757|6552|125F|50DD|6363|6365|6A7A|A911|4400|A112|C563|C3A1|E911|E1C1|E411|E123|C350|C4D4|C3C3|C3B0|C737|C3A1|C549|C6A7"`
to checked all the spyfiles in https://github.com/walczakp/rds-spy-logs for ODA other than RT+ and TMC.

# RDS2 ODA (hex 6365)
https://github.com/walczakp/rds-spy-logs/blob/master/Czech%20Republic/2337%20-%202019-05-04%2016-26-20.spy
```
Czech Republic/2337 - 2019-05-04 16-26-20.spy:55:2337 2728 6C61 6365 @2019/05/04 16:26:24.32
Czech Republic/2337 - 2019-05-04 16-26-20.spy:119:2337 2728 6C61 6365 @2019/05/04 16:26:29.95
```
and
https://github.com/walczakp/rds-spy-logs/blob/master/Germany/D312%20-%202019-05-04%2015-21-32.spy
```
Germany/D312 - 2019-05-04 15-21-32.spy:434:D312 20F6 7669 6365 @2019/05/04 15:22:09.64
Germany/D312 - 2019-05-04 15-21-32.spy:530:D312 20F6 7669 6365 @2019/05/04 15:22:18.04
```
and
https://github.com/walczakp/rds-spy-logs/blob/master/Germany/D6F1%20-%202019-05-04%2015-08-15.spy
```
Germany/D6F1 - 2019-05-04 15-08-15.spy:30:D6F1 2402 7669 6365 @2019/05/04 15:08:17.48
Germany/D6F1 - 2019-05-04 15-08-15.spy:53:D6F1 2402 7669 6365 @2019/05/04 15:08:19.48
```
and
https://github.com/walczakp/rds-spy-logs/blob/master/Poland/3F44%20-%202017-08-13%2000-14-01%20PL%20LODZ%20RMF.rds
```
Poland/3F44 - 2017-08-13 00-14-01 PL LODZ RMF.rds:631:3F44 054A 3C72 6365 @2017/08/13 00:14:58.197
Poland/3F44 - 2017-08-13 00-14-01 PL LODZ RMF.rds:643:3F44 054A 233C 6365 @2017/08/13 00:14:59.269
```
# Warning receiver (hex 6A7A)
https://github.com/walczakp/rds-spy-logs/blob/master/France/F201%20-%202018-01-02%2019-10-59.spy
```
France/F201 - 2018-01-02 19-10-59.spy:1032:F201 8408 47EC 6A7A @2018/01/02 19:12:21.68
France/F201 - 2018-01-02 19-10-59.spy:1036:F201 8408 47EC 6A7A @2018/01/02 19:12:22.03
```
# ELECTRABEL-DSM 20 (hex 1C5E)
https://github.com/walczakp/rds-spy-logs/blob/master/UK/C36C%20-%202015-09-27%2023-29-22%20UK%20NRW%20HEART.rds
```
UK/C36C - 2015-09-27 23-29-22 UK NRW HEART.rds:91:C36C 8548 0883 1C5E @2015/09/27 23:29:33.456
UK/C36C - 2015-09-27 23-29-22 UK NRW HEART.rds:97:C36C 8548 0883 1C5E @2015/09/27 23:29:33.984
```
# Cross referencing DAB within RDS (hex 0093)
https://github.com/walczakp/rds-spy-logs/blob/master/UK/C204%20-%202015-09-27%2023-35-46%20UK%20NRW%20BBC4.rds
```
UK/C204 - 2015-09-27 23-35-46 UK NRW BBC4.rds:9336:C204 3138 0000 0093 @2015/09/27 23:49:23.896
UK/C204 - 2015-09-27 23-35-46 UK NRW BBC4.rds:10028:C204 3138 0000 0093 @2015/09/27 23:50:24
```
# ELECTRABEL-DSM 12 (hex E319)
https://github.com/walczakp/rds-spy-logs/blob/master/Russia/7205%20-%202019-05-04%2001-18-20.spy
```
Russia/7205 - 2019-05-04 01-18-20.spy:1305:7205 852D 1323 E319 @2019/05/04 01:20:14.64
Russia/7205 - 2019-05-04 01-18-20.spy:1309:7205 852D 1323 E319 @2019/05/04 01:20:14.98
```
# Personal weather station (hex 5757)
https://github.com/walczakp/rds-spy-logs/blob/master/Russia/7AA1%20-%202019-05-04%2001-27-45.spy
```
Russia/7AA1 - 2019-05-04 01-27-45.spy:174:7AA1 01C8 E0CD 5757 @2019/05/04 01:28:00.41
Russia/7AA1 - 2019-05-04 01-27-45.spy:181:7AA1 01C8 E0CD 5757 @2019/05/04 01:28:01.01
```
# Enhanced early warning system (hex 7373)
https://github.com/walczakp/rds-spy-logs/blob/master/Russia/7261%20-%202019-05-04%2001-16-40.spy
```
Russia/7261 - 2019-05-04 01-16-40.spy:11:7261 2520 5261 7373 @2019/05/04 01:16:37.69
Russia/7261 - 2019-05-04 01-16-40.spy:39:7261 2520 5261 7373 @2019/05/04 01:16:40.16
```
and
https://github.com/walczakp/rds-spy-logs/blob/master/USA/7DC9%20-%202019-05-04%2021-51-15.spy
```
USA/7DC9 - 2019-05-04 21-51-15.spy:7:7DC9 34F8 005E 7373 @2019/05/04 21:51:13.99
USA/7DC9 - 2019-05-04 21-51-15.spy:14:7DC9 34F8 0096 7373 @2019/05/04 21:51:14.62
```
[... 36 lines cut ...]
and
https://github.com/walczakp/rds-spy-logs/blob/master/USA/8FC4%20-%202019-05-04%2021-55-19.spy
```
USA/8FC4 - 2019-05-04 21-55-19.spy:27:8FC4 3556 A08F C3B0 @2019/05/04 21:55:11.03
USA/8FC4 - 2019-05-04 21-55-19.spy:75:8FC4 3556 A08F C3B0 @2019/05/04 21:55:15.25
```
and
https://github.com/walczakp/rds-spy-logs/blob/master/USA/4569%20-%202020-08-19%2020-45-06.spy
(see PR)
```
USA/4569 - 2020-08-19 20-45-06.spy:8:4569 30D6 A045 C3B0 @2020/08/19 20:45:05.70
USA/4569 - 2020-08-19 20-45-06.spy:51:4569 ---- A045 C3B0 @2020/08/19 20:45:09.39
```
# RDS-Light (hex 4400) 
https://github.com/walczakp/rds-spy-logs/blob/master/Finland/6403%20-%202018-09-05%2015-15-59%20FI%20HE%20YLESUOMI.rds
```
Finland/6403 - 2018-09-05 15-15-59 FI HE YLESUOMI.rds:65:6403 8440 182B 4400 @6884
Finland/6403 - 2018-09-05 15-15-59 FI HE YLESUOMI.rds:264:6403 8440 182B 4400 @7580
```
# CITIBUS 2 (hex 4C59)
https://github.com/walczakp/rds-spy-logs/blob/master/Germany/D52F%20-%202018-11-01%2014-28-26%20DE%20BER%20SPREE.rds
```
Germany/D52F - 2018-11-01 14-28-26 DE BER SPREE.rds:5024:D52F 254A 494C 4C59 @2018/11/01 14:38:52.461
Germany/D52F - 2018-11-01 14-28-26 DE BER SPREE.rds:5061:D52F 254A 494C 4C59 @2018/11/01 14:38:55.693
```

windytan, 2020-08-21:
Thanks a lot! That looks like a very useful repo. I think I'll have Redsea print out the raw ODA data in case the application is not supported, since some of these are unpublished.

windytan, 2020-08-21:
Running all of these logs through redsea it seems that there are actual ODA groups for only iTunes tagging (USA PI codes 0x8FC4 and 0x4569) and DAB cross-referencing (UK PI codes 0xC204 and 0xC202). The rest are probably spurious errors or otherwise coincidental, but I'll leave the issue open in case there's more info about the iTunes protocol.

andimik, 2023-07-03:
I found a list of most common ODAs also in the manual of a professional RDS encoder seller:

Page 22 on https://pira.cz/fm_broadcast_analyzer/p75-p175-manual.pdf

andimik, 2024-02-20:
There is a Finnish program using 

# RadioText Plus / RT+ for eRT (dec 9416, hex 0x4BD8) 

# Enhanced RadioText / eRT (dec 25938, hex 0x6552) 

```
$ cat Downloads/rds2-mpx-6255.wav | redsea -r 192k | grep oda | sort | uniq
{"pi":"0x6255","group":"12A","prog_type":"Varied","tp":true,"unknown_oda":{"app_name":"Enhanced RadioText (eRT)","raw_data":"00 4AC3 A472"}}
{"pi":"0x6255","group":"12A","prog_type":"Varied","tp":true,"unknown_oda":{"app_name":"Enhanced RadioText (eRT)","raw_data":"01 7669 7261"}}
{"pi":"0x6255","group":"12A","prog_type":"Varied","tp":true,"unknown_oda":{"app_name":"Enhanced RadioText (eRT)","raw_data":"02 6469 6F20"}}
{"pi":"0x6255","group":"12A","prog_type":"Varied","tp":true,"unknown_oda":{"app_name":"Enhanced RadioText (eRT)","raw_data":"03 5244 5332"}}
{"pi":"0x6255","group":"12A","prog_type":"Varied","tp":true,"unknown_oda":{"app_name":"Enhanced RadioText (eRT)","raw_data":"04 2045 5254"}}
{"pi":"0x6255","group":"12A","prog_type":"Varied","tp":true,"unknown_oda":{"app_name":"Enhanced RadioText (eRT)","raw_data":"05 0D0D 0D0D"}}
{"pi":"0x6255","group":"12A","prog_type":"Varied","tp":true,"unknown_oda":{"raw_data":"00 4AC3 A472"}}
{"pi":"0x6255","group":"12A","prog_type":"Varied","tp":true,"unknown_oda":{"raw_data":"01 7669 7261"}}
{"pi":"0x6255","group":"12A","prog_type":"Varied","tp":true,"unknown_oda":{"raw_data":"02 6469 6F20"}}
{"pi":"0x6255","group":"12A","prog_type":"Varied","tp":true,"unknown_oda":{"raw_data":"03 5244 5332"}}
{"pi":"0x6255","group":"15A","prog_type":"Varied","tp":true,"unknown_oda":{"raw_data":"00 4AC3 A452"}}
{"pi":"0x6255","group":"15A","prog_type":"Varied","tp":true,"unknown_oda":{"raw_data":"01 5649 5241"}}
{"pi":"0x6255","group":"15A","prog_type":"Varied","tp":true,"unknown_oda":{"raw_data":"02 4449 4F0D"}}
{"pi":"0x6255","group":"15A","prog_type":"Varied","tp":true,"unknown_oda":{"raw_data":"03 0D0D 0D0D"}}
{"pi":"0x6255","group":"3A","debug":["TODO: Unimplemented ODA app 19416"],"open_data_app":{"app_name":"RadioText Plus / RT+for eRT","message":0,"oda_group":"13A"},"prog_type":"Varied","tp":true}
{"pi":"0x6255","group":"3A","debug":["TODO: Unimplemented ODA app 25938"],"open_data_app":{"app_name":"Enhanced RadioText (eRT)","message":1,"oda_group":"12A"},"prog_type":"Varied","tp":true}
```

See https://www.rundfunkforum.de/viewtopic.php?p=1749186#p1749186

closed 2024-08-14 by windytan

windytan, 2025-10-15:
Awesome field research andimik!