windytan/redsea #76: Redsea 0.20 on Enigma2 OpenATV images stopped giving output after latest update

MCelliotG, 2021-08-18:
As of yesterday's update, something has severely affected the redsea output on openATV 6.4 images and rds is not working anymore, using an RTL2832 based USB receiver.
Here's the output from rtl_fm 
root@mutant51:~# rtl_fm -f 90.80M -M fm -l 0 -A std -s 171k -g 8 -p -99 -F 9 - | redsea -e
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to 7.70 dB.
Tuner error set to -99 ppm.
Tuned to 91142000 Hz.
Oversampling input by: 8x.
Oversampling output by: 1x.
Buffer size: 5.99ms
Exact sample rate is: 1368000.013046 Hz
Sampling at 1368000 S/s.
Output at 171000 Hz.

windytan, 2021-08-19:
Hi, thanks for the report. I have some questions as I haven't heard about OpenATV before. Is redsea used there? What was updated yesterday? Do you mean updated to redsea 0.20 from 0.19?

MCelliotG, 2021-08-19:
Yes, redsea is used inside the SDGradio plugin for Enigma2 images. Yes, I will record a segment, could you remind me the command please?

windytan, 2021-08-20:
`rtl_fm -f 90.80M -M fm -l 0 -A std -s 171k -g 8 -p -99 -F 9 > output.pcm`

After it has run a few seconds, press Ctrl-C so it stops, otherwise the file can get quite large fast.

MCelliotG, 2021-08-20:
Thank you, here is the output.
https://wetransfer.com/downloads/a35aac7af86d58530fe542b8436c477920210820104235/b8f926

andimik, 2021-08-20:
Well, it works in general with your file (I played it on my laptop), but the bit error rate is very high. 

Please try to find a better (local) Greek station other than ZOO. Because this station has a rotating PS (according to fmlist.org)

```ZOO_90.8 __HIT___ __MUSIC_ __ONLY__ __NOW___ _ON_AIR: [Scrolling_artist_and_title]```

But I fear it has nothing to do with redsea itself. Maybe you discuss better in https://www.opena.tv/ if others have the same (negative) experience.

Or use a better antenna and a difference radio station.

Of course you can start redsea on the set top box load this file `output.pcm` described in the syntax below.

```
$ cat Downloads/output.pcm | redsea -E
{"bler":6,"pi":"0x10C7"}
{"bler":10,"di":{"dynamic_pty":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":83,"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":85,"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":81,"di":{"dynamic_pty":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":79,"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"bler":75,"pi":"0x10C7"}
{"bler":68,"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":66,"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":66,"pi":"0x10C7"}
{"bler":66,"pi":"0x10C7"}
{"bler":68,"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":75,"di":{"dynamic_pty":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":81,"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"bler":85,"pi":"0x10C7","prog_type":"Country music","tp":true}
{"bler":89,"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":87,"pi":"0x10C7"}
{"bler":77,"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"bler":79,"pi":"0x10C7"}
{"bler":79,"pi":"0x10C7"}
{"bler":70,"pi":"0x10C7"}
{"bler":72,"pi":"0x10C7"}
{"bler":75,"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"bler":77,"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":79,"pi":"0x10C7"}
{"bler":81,"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":77,"pi":"0x10C7"}
{"bler":77,"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"bler":75,"pi":"0x10C7"}
{"bler":83,"pi":"0x10C7"}
{"bler":81,"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"bler":79,"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"bler":70,"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":68,"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":68,"pi":"0x10C7"}
{"bler":70,"di":{"dynamic_pty":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":70,"pi":"0x10C7"}
{"bler":70,"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":83,"pi":"0x10C7"}
{"bler":79,"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"bler":79,"pi":"0x10C7"}
{"bler":75,"pi":"0x10C7"}
{"bler":75,"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":75,"pi":"0x10C7"}
{"bler":77,"pi":"0x10C7"}
{"bler":77,"di":{"dynamic_pty":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":81,"pi":"0x10C7"}
{"bler":81,"pi":"0x10C7"}
{"bler":83,"pi":"0x10C7"}
{"bler":81,"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"bler":81,"pi":"0x10C7"}
{"bler":87,"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":83,"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":83,"di":{"compressed":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":77,"pi":"0x10C7"}
{"bler":75,"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"bler":72,"di":{"stereo":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":66,"di":{"artificial_head":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":60,"di":{"dynamic_pty":false},"group":"0A","is_music":true,"pi":"0x10C7","prog_type":"Pop music","ta":false,"tp":true}
{"bler":62,"group":"2A","pi":"0x10C7","prog_type":"National music","tp":false}
{"bler":60,"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
{"bler":64,"group":"2A","pi":"0x10C7","prog_type":"Pop music","tp":true}
```

This is the Windows software (freeware) rdsspy (see http://rdsspy.com/ ) via Wine

[image]

You see lots of garbage ...

[output.zip](https://github.com/windytan/redsea/files/7023912/output.zip)

MCelliotG, 2021-08-21:
The problem is not the station and whatever errors, it's the complete lack of output. Since redsea 0.20 decodes the rds on other systems, it looks like the problem is with the Enigma2 update and the libraries that the SDGradio depends on such as libsnd. I will try to manually replace the libraries on the latest openatv image and report back.

MCelliotG, 2021-08-21:
OK, I just got to the bottom of this. No problem with redsea 0.20. The issue was with liquid-dsp used in the SDGradio. OpenATV upgraded the version to 1.3. Downgrading to version 1.0 fixed the RDS output immediately.
You can consider the issue fixed.

windytan, 2021-08-21:
That's interesting, it should work with 1.3. Glad you found the root cause nevertheless. By the way, if the block error rate ever becomes a problem, you could try adjusting the -p value to `-p -6` for this receiver unit to get cleaner data.

MCelliotG, 2021-08-21:
That's a good tip, thank you so much.
No clue why liquid-dsp affects the output so severely. I will talk about it with the plugin creators.

MCelliotG, 2021-08-21:
BTW, here's another issue. In this station radiotext is not decoded by redsea, only partial_radiotext. Could you check it out?
The station surely transmits radiotext, but with redsea it's only visible with the partial parameter.
https://we.tl/t-TWS81IsPB1

windytan, 2021-08-21:
The station doesn't seem to transmit the radiotext end marker. Redsea doesn't know that the message ends at 24 characters, and never prints the message. This type of situation is tricky to detect but could be possible with a few changes. Would be interesting to know how actual RDS receivers deal with this station.

MCelliotG, 2021-08-21:
But if you run redsea with the -p command you will see that radiotext is transmitted properly. Is it possible that they have radiotext in a block redsea does not recognize?

windytan, 2021-08-22:
I'll open a new issue, just for tracking.

closed 2021-08-22 by windytan