windytan/redsea #78: redsea no longer compiles due to changes in liquid-dsp

WanWizard, 2021-11-01:
modem was changed to modemcf.
````
| In file included from ../src/subcarrier.h:29,
|                  from subcarrier.cc:17:
| ../src/liquid_wrappers.h:99:3: error: 'modem' does not name a type; did you mean 'Modem'?
|    99 |   modem object_;
|       |   ^~~~~
|       |   Modem
| In file included from ../src/subcarrier.h:29,
|                  from redsea.cc:26:
| ../src/liquid_wrappers.h:99:3: error: 'modem' does not name a type; did you mean 'Modem'?
|    99 |   modem object_;
|       |   ^~~~~
|       |   Modem
````

windytan, 2021-11-03:
Thanks for the heads-up!

acmahalkasi, 2021-11-10:
I still get that error. have you found a proper way to fix?

I install liquid-dsp and use Linux ubuntu

closed 2021-11-13 by windytan