windytan/redsea #79: rtl_sdr output to redsea

acmahalkasi, 2021-12-01:
Hello everyone,
I have a couple of questions. I hope anybody has an idea.

1) Trying to extract RDS data using redsea from rtl_sdr command which produces IQ data as far as I know. 

Command was "rtl_sdr -f 91400000 -s 2500000 -n 25000000 voice.dat"
Is there any way to make that?

2) I changed the rtl_fm code to be reading IQ pairs from a file which is saved before running the rtl_sdr command. My application (I will say "edited rtl_fm" instead of my application) is taking the file, which has been recorded via rtl_sdr and demodulating as same as original rtl_fm. So actually edited rtl_fm doesn't use sdr-card anymore because it doesn't need it. If I use the original rtl_fm, redsea is working, but if I edited rtl_fm redsea is not able to extract any data. Do we have to use sdr card properties to do that?

Command "./rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 20 -F 9 -f 91400000 | redsea -p" is working
Command "./rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 20 -F 9 -f 91400000 -i voice.dat | redsea -p" is not working

Thanks for your valuable time.

windytan, 2021-12-01:
Hi,
1. You could use, for example, [csdr](https://github.com/ha7ilm/csdr) to demodulate FM from an IQ file or stream. Some tips here https://github.com/windytan/redsea/wiki/Use-cases
2. A physical SDR is not needed, any MPX stream should work. I have two requests: Firstly, could you upload a few seconds worth of output from the edited rtl_fm somewhere? I could take a look if something is wrong with the RDS subcarrier. Secondly, what's the output of`redsea --version` ?

acmahalkasi, 2021-12-01:
Hi, 
1) Using rtl_sdr, which is recording IQ stream, is more preferable right now, yet I will look at csdr instead of rtl_fm to demodulate the data and give information about the results soon. In addition, we managed to demodulate and listen to the output of the edited rtl_fm as I mentioned in Question 2.

2) Output of the "redsea --version" is "redsea 0.20 by OH2EIQ"
Adding the link of the input and the output.
Output of edited rtl_fm : https://drive.google.com/file/d/1hFGQdgfpqxtZ6ApceZNotm_JWnhlltou/view?usp=sharing
Input:  https://drive.google.com/file/d/1xfKURjMrakldpCP2YvWVzjZio3SMzcXU/view?usp=sharing

windytan, 2021-12-01:
Based on the pilot tone it seems that the samplerate of voice.raw is 312.5 kHz (which would be 2.5 MHz divided by 8000). Try `redsea -r 312500` (though this particular voice.raw might be too noisy).

0m3rF, 2021-12-01:
Hi @windytan, thanks for your reply. 

We are trying with acmahalkasi to listen local radios as many as possible. Also we need station's RDS information. Given rtl_sdr command must have 12 radio channels because of sample rate (for every 200khz in 2.5M ). 

With edited rtl_fm command we can get just 1 station's broadcast and it has good sound quality. We probably couldn't managed to shift or decimate to get different stations. It is possible that we are giving all channels data to redsea so we can't get any RDS.

We tried redsea's sample_rate option but we got no output. We tried csdr as you suggested but it didn't work either also its sound is so noisy. 

We have one SDR dongle so recording IQ data with rtl_sdr is important for us. We will process IQ data later for every station to get audio and RDS. 

How do you think vioce.raw become 312.5 hz? we should have 12 different frequencies. why divided to 8 ?

windytan, 2021-12-01:
Unfortunately I'm not familiar with the internals of rtl_fm or rtl-sdr, so I don't know why it could be 312.5 kHz. It shouldn't matter if you give the `-r` option to redsea. But it could be problematic if interference from adjacent channels leaks in, because of the inadequate pre-decimation filtering. RDS is more sensitive to interference than the audio.

closed 2021-12-05 by acmahalkasi