windytan/redsea #82: JSON output of TMC location table with Unicode characters

andimik, 2022-05-22:
After several years I have tried TMC again today and discovered that the JSON output prints the C++ Unicode source code and not the character like in `NAMES.DAT`

For example:

# Slovenia

```
$ rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 20 -F 9 -f 104.0M | redsea -E -l /media/andreas/Data/ut/TMC/SVN/ | fgrep '\'
[...]
{"pi":"0x9202","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"coordinates":[{"lat":46.25888,"lon":14.67968}],"description":"Single alternate line traffic.","direction":"both","event_codes":[513],"extent":"+0","location":31082,"road_name":"Menge\u0161 - Braslov\u010de","road_number":"225","update_class":5,"urgency":"none"}},"tp":true}
{"pi":"0x9202","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"coordinates":[{"lat":46.03547,"lon":14.84292}],"description":"Single alternate line traffic.","direction":"both","event_codes":[513],"extent":"+0","location":41113,"road_name":"Litija - Bi\u010d","road_number":"416","update_class":5,"urgency":"none"}},"tp":true}
{"pi":"0x9202","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"coordinates":[{"lat":46.41153,"lon":15.79268}],"description":"Closed due to roadworks.","direction":"both","event_codes":[735],"extent":"+0","location":33794,"road_name":"Fram - Kidri\u010devo","road_number":"711","update_class":5,"urgency":"U"}},"tp":true}
```
Hint: Menge\u0161 - Braslov\u010de is Mengeš - Braslovče 

# Austria

```
$ rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 20 -F 9 -f 90.4M | redsea -E -l /media/andreas/Data/ut/TMC/AUT/AT_103_E0_A_1_3.5_LTEF | fgrep '\'
[...]
{"pi":"0xA203","group":"8A","bler":0,"prog_type":"Pop music","tmc":{"message":{"coordinates":[{"lat":46.62345,"lon":13.90805},{"lat":46.62485,"lon":13.92305},{"lat":46.6217,"lon":13.94266}],"description":"Delays of up to 5 minutes.","direction":"both","event_codes":[1601],"extent":"+2","location":49031,"road_name":"K\u00e4rntner Stra\u00dfe","road_number":"B83","span_from":"Zauchen","span_to":"A2 - Wernberg","update_class":20,"urgency":"none"}},"tp":true}
{"pi":"0xA203","group":"8A","bler":0,"prog_type":"Pop music","tmc":{"message":{"coordinates":[{"lat":48.0055,"lon":13.67472}],"description":"Closed due to roadworks.","direction":"both","event_codes":[735],"extent":"-0","location":42179,"road_name":"Hausruck Stra\u00dfe","road_number":"B143","update_class":5,"urgency":"U"}},"tp":true}
{"pi":"0xA203","group":"8A","bler":0,"prog_type":"Pop music","tmc":{"message":{"coordinates":[{"lat":47.79785,"lon":13.398},{"lat":47.81628,"lon":13.36415}],"description":"Single alternate line traffic. Construction work.","direction":"single","event_codes":[513,803],"extent":"+1","location":36918,"road_name":"Mondsee Stra\u00dfe","road_number":"B154","span_from":"Ober\u00f6sterreich / Salzburg","span_to":"Plomberg","update_class":5,"urgency":"none"}},"tp":true}
```
Hint: K\u00e4rntner Stra\u00dfe is Kärntner Straße

This seems to be in redsea/src/tmc/locationdb.cc lines 65ff.

windytan, 2022-05-22:
Thanks! Just for reference, I think I need to embed a newer version of jsoncpp and enable emitUTF8 https://stackoverflow.com/questions/50637362/how-can-i-ask-jsoncpp-to-encode-unicode-strings

windytan, 2022-05-23:
See if this helps! I quickly tested by converting one of the location tables I have into UTF-8, and it printed the names fine.

andimik, 2022-05-23:
Yes, this works now :-)

```
$ rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 20 -F 9 -f 90.4M | redsea -E -l /media/andreas/Data/ut/TMC/AUT/ | grep ß
[...]
{"pi":"0xA203","group":"8A","bler":2,"prog_type":"Pop music","tmc":{"message":{"coordinates":[{"lat":47.79785,"lon":13.398},{"lat":47.81628,"lon":13.36415}],"description":"Single alternate line traffic. Construction work.","direction":"single","event_codes":[513,803],"extent":"+1","location":36918,"road_name":"Mondsee Straße","road_number":"B154","span_from":"Oberösterreich / Salzburg","span_to":"Plomberg","update_class":5,"urgency":"none"}},"tp":true}
{"pi":"0xA203","group":"8A","bler":0,"prog_type":"Pop music","tmc":{"message":{"coordinates":[{"lat":47.40637,"lon":12.9732}],"description":"Single alternate line traffic. Construction work.","direction":"single","event_codes":[513,803],"extent":"-0","location":31015,"road_name":"Hochkönig Straße","road_number":"B164","update_class":5,"urgency":"none"}},"tp":true}
{"pi":"0xA203","group":"8A","bler":0,"prog_type":"Pop music","tmc":{"message":{"coordinates":[{"lat":46.56004,"lon":13.27568}],"description":"Closed. Construction work.","direction":"both","event_codes":[401,803],"extent":"-0","location":6723,"road_name":"Nassfeld Straße","road_number":"B90","update_class":5,"urgency":"U"}},"tp":true}
```

```
$ rtl_fm -M fm -l 0 -A std -p 0 -s 171k -g 20 -F 9 -f 104M | redsea -E -l /media/andreas/Data/ut/TMC/SVN/ | grep š
[...]
{"pi":"0x9202","group":"8A","bler":4,"prog_type":"No PTY","tmc":{"message":{"coordinates":[{"lat":46.48195,"lon":15.65783}],"description":"Slip road restrictions.","direction":"single","event_codes":[409],"extent":"-0","location":30136,"road_name":"Maribor - Gruškovje","road_number":"A4","update_class":7,"urgency":"none"}},"tp":true}
{"pi":"0x9202","group":"8A","bler":17,"prog_type":"No PTY","tmc":{"message":{"coordinates":[{"lat":45.93114,"lon":14.06465}],"description":"Single alternate line traffic.","direction":"both","event_codes":[513],"extent":"+0","location":44444,"road_name":"Godovič - Ajdovščina","road_number":"207","update_class":5,"urgency":"none"}},"tp":true}
```

I'll close this issue.

closed 2022-05-23 by andimik