windytan/redsea #88: AF shown too less in JSON even for error-free signals

andimik, 2023-05-12:
I made several experiments with various RDS signals, but the alternative frequency list (AF in group 0A) is shown only a few times in the JSON, whereas the EON frequencies (group 14A) are displayed regularly.

The help says,
```
Under noisy conditions, redsea may not be able to 
fully receive all information. Multi-group data
such as PS names, RadioText, and alternative
frequencies are especially vulnerable. This option
makes it display them even if not fully received,
as partial_{ps,radiotext,alt_kilohertz}.
```
When I enable `-p` they will appear quite often, but I don't think this makes sense here as in my case, the signal is error-free nearly all the time (parameter `-E`)

[samples.zip](https://github.com/windytan/redsea/files/11468180/samples.zip)

for example 

`cat ./A201_2023-05-13.spy | redsea -hE | grep kilo | grep 0A`
resp.
`cat ./A201_2023-05-13.spy | redsea -hE | grep kilo | grep 14A`
or
`cat ./A201_2023-05-13.spy | redsea -hE | sort | uniq -c | grep kilo`

```
$ cat ./A203_2023-05-13.spy | redsea -hE | sort | uniq -c | grep kilo
      1 {"pi":"0xA203","group":"0A","alt_kilohertz":[87900,88000,88100,88400,88600,88700,88800,89100,89200,89600,90400,97000,98300,98400,98700,99000,99500,99800,100700,102500,102600,103100,103300],"bler":0,"di":{"compressed":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
     14 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":87600,"tp":false},"prog_type":"Pop music","tp":true}
     15 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":87700,"tp":false},"prog_type":"Pop music","tp":true}
     10 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":88200,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":89000,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":89100,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":89300,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":89500,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":89700,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":89800,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":90000,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":90300,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":90700,"tp":false},"prog_type":"Pop music","tp":true}
     15 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":90900,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":91000,"tp":false},"prog_type":"Pop music","tp":true}
     15 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":91100,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":91200,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":91300,"tp":false},"prog_type":"Pop music","tp":true}
     10 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":91400,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":91500,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":91600,"tp":false},"prog_type":"Pop music","tp":true}
     10 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":91800,"tp":false},"prog_type":"Pop music","tp":true}
      4 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":91900,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":92000,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":92100,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":92400,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":92800,"tp":false},"prog_type":"Pop music","tp":true}
     10 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":93200,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":93600,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":93900,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":94300,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":96700,"tp":false},"prog_type":"Pop music","tp":true}
     10 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA201","kilohertz":98800,"tp":false},"prog_type":"Pop music","tp":true}
      4 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA213","kilohertz":100300,"tp":false},"prog_type":"Pop music","tp":true}
      4 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA213","kilohertz":101000,"tp":false},"prog_type":"Pop music","tp":true}
      4 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA213","kilohertz":101700,"tp":false},"prog_type":"Pop music","tp":true}
      4 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA213","kilohertz":101900,"tp":false},"prog_type":"Pop music","tp":true}
      4 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA213","kilohertz":102100,"tp":false},"prog_type":"Pop music","tp":true}
      4 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA213","kilohertz":102300,"tp":false},"prog_type":"Pop music","tp":true}
      4 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA213","kilohertz":102900,"tp":false},"prog_type":"Pop music","tp":true}
      4 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA213","kilohertz":103600,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA502","kilohertz":100400,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA502","kilohertz":100900,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA502","kilohertz":89300,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA502","kilohertz":91500,"tp":false},"prog_type":"Pop music","tp":true}
[... 203 lines cut ...]
     10 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA502","kilohertz":95500,"tp":false},"prog_type":"Pop music","tp":true}
     15 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA502","kilohertz":95600,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA502","kilohertz":95700,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA502","kilohertz":95800,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA502","kilohertz":95900,"tp":false},"prog_type":"Pop music","tp":true}
      5 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA502","kilohertz":96000,"tp":false},"prog_type":"Pop music","tp":true}
[... cut ...]
      4 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA902","kilohertz":95900,"tp":false},"prog_type":"Pop music","tp":true}
      4 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA902","kilohertz":96300,"tp":false},"prog_type":"Pop music","tp":true}
      4 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA902","kilohertz":96800,"tp":false},"prog_type":"Pop music","tp":true}
      4 {"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA902","kilohertz":98100,"tp":false},"prog_type":"Pop music","tp":true}
```

windytan, 2023-05-15:
Hi, thanks again for bug-hunting redsea!

I believe redsea is not properly recognizing AF method B here. In Method B, several different lists are sent, corresponding to AFs of different tuned frequencies. The length of the list is encoded differently in these methods, which causes this weird behavior. This needs to be fixed for sure.

andimik, 2023-06-12:
I've tried that, but I fear we get too much results now (see below txt file). 

Due to the fact that the number of AFs (including the own) for a particular radio tower is transmitted (in the example below 9 resp. 11) I expect to wait until all data have arrived (= not partial) and then print

a) the number of frequencies (see below) and then
b) a sorted list of frequencies of the same program and then 
c) a sorted list of regional variant, maybe marked with an `R`

like that:

```
"alt_b_kilohertz":[89300:88800,99500,101700,102600R,89000R]
"alt_b_kilohertz":[99500:89300,99500,104800R,89100R]
```

`cat ./A203_2023-05-13.spy | redsea -hEp  | grep alt_kilo | uniq -c | sort > /tmp/A203_2023-05-13.spy_af_groupb.txt`

[A203_2023-05-13.spy_af_groupb.txt](https://github.com/windytan/redsea/files/11728314/A203_2023-05-13.spy_af_groupb.txt)

from IEC 62106 page 45

> # Examples of an AF method B coding:
> 
> F1 | F2 | Commentary
> --- | --- | ---
>   |   |  
> \# 11 | 89.3 | Total number (11) of frequencies for tuning frequency (89.3)
> 89.3 | 99.5 | F2 > F1 hence 99.5 is an AF of tuned frequency 89.3, and is the same programme
> 89.3 | 101.7 | F2 > F1 hence 101.7 is an AF of tuned frequency 89.3, and is the same programme
> 88.8 | 89.3 | F2 > F1 hence 88.8 is an AF of tuned frequency 89.3, and is the same programme
> 102.6 | 89.3 | F2 < F1 hence 102.6 is an AF of a regional variant of tuned frequency 89.3
> 89.3 | 89.0 | F2 < F1 hence 89.0 is an AF of a regional variant of tuned frequency 89.3
>   |   |  
> \# 9 | 99.5 | Total number (9) of frequencies for tuning frequency (99.5)
> 89.3 | 99.5 | F2 < F1 hence 89.3 is an AF of tuned frequency 99.5, and is the same programme
> 99.5 | 100.9 | F2 > F1 hence 100.9 is an AF of tuned frequency 99.5, and is the same programme
> 104.8 | 99.5 | F2 < F1 hence 104.8 is an AF of a regional variant of tuned frequency 99.5
> 99.5 | 89.1 | F2 < F1 hence 89.1 is an AF of a regional variant of tuned frequency 99.5

What do you think?

andimik, 2023-06-13:
This is great!

```
$ cat A203_-_2023-06-09_09-15-35.spy | redsea -hE | jq -c ".alt_frequencies_b" | sort | uniq
null
{"tuned_frequency":100700,"same_programme":[88000,89100,90400,102600]}
{"tuned_frequency":101200,"same_programme":[90400,98100,100900]}
{"tuned_frequency":102500,"same_programme":[88800,90400]}
{"tuned_frequency":102600,"same_programme":[100700]}
{"tuned_frequency":103300,"same_programme":[88600,89100,90400,98400]}
{"tuned_frequency":107000,"same_programme":[90400,97000]}
{"tuned_frequency":88000,"same_programme":[88400,88700,89100,89600,90400,98300,98700,99000,99800,100700,103100]}
{"tuned_frequency":88000,"same_programme":[98600,99500]}
{"tuned_frequency":88100,"same_programme":[90400]}
{"tuned_frequency":88600,"same_programme":[103300]}
{"tuned_frequency":88700,"same_programme":[88000,89600,90400]}
{"tuned_frequency":88700,"same_programme":[90400]}
{"tuned_frequency":88800,"same_programme":[90400,99900,102500]}
{"tuned_frequency":88900,"same_programme":[89200,98700,99500]}
{"tuned_frequency":89100,"same_programme":[88000,89600,90400,90700,99300,100700,103300]}
{"tuned_frequency":89600,"same_programme":[88000,88500,88700,89100,90400,90700,99300,99500]}
{"tuned_frequency":90400,"same_programme":[87900,88000,88100,88800,89100,89200,89600,97000,98300,98400,99500,102500]}
{"tuned_frequency":90400,"same_programme":[88600,88700,89500,90700,98600,98800,99300,100700,101200,103300,107000]}
{"tuned_frequency":90700,"same_programme":[89100,89600,90400,99300]}
{"tuned_frequency":97000,"same_programme":[90400,101200,107000]}
{"tuned_frequency":98200,"same_programme":[87900]}
{"tuned_frequency":98300,"same_programme":[88000,88400,99800]}
{"tuned_frequency":98300,"same_programme":[90400,99500]}
{"tuned_frequency":98400,"same_programme":[88600,90400,99300,103300]}
{"tuned_frequency":98600,"same_programme":[88000,90400]}
{"tuned_frequency":98600,"same_programme":[99500]}
{"tuned_frequency":99300,"same_programme":[89100,89600,90400,90700]}
{"tuned_frequency":99300,"same_programme":[90400,98400]}
{"tuned_frequency":99500,"same_programme":[88000,88900,89200,89600,90400,98300,98600]}
{"tuned_frequency":99800,"same_programme":[87900,89700]}
{"tuned_frequency":99900,"same_programme":[88800]}
```

With this method a technician can check if the AF list is complete or needs to be fixed.

Or this one:

```
$ cat A3FF_-_2021-07-18_18-56*.spy | redsea -h | jq -c ".alt_frequencies_b" | grep -v null | sort | uniq
{"tuned_frequency":103700,"same_programme":[91600,98400,100800,103400,105300,106200,107600],"regional_variants":[101900]}
{"tuned_frequency":105300,"same_programme":[91600,103400,103700,106200,107600],"regional_variants":[101900]}
{"tuned_frequency":106200,"same_programme":[103700,105300,106600,107600],"regional_variants":[101900]}
{"tuned_frequency":106600,"same_programme":[91600,100800,103700,106200,107600]}
{"tuned_frequency":91600,"same_programme":[103400,103700,105300,106600,107600]}
```

**That's what I have expected.** In my mind I can see a map just looking at this JSON output :-) Because I know where (on which site) all these frequencies are transmitted and which AF is the neighbor transmitting site.

For example this one (source: FMLIST)

[image]

> I'm just wondering whether sorting the list is the way to go. There might be some importance sorting done by the broadcaster that is lost if we sort numerically.

Yes, this is the right way in my opinion.

[A3FF_-_2021-07-18_18-5.zip](https://github.com/windytan/redsea/files/11740050/A3FF_-_2021-07-18_18-5.zip)

Because RDSSpy's log unnecessarily repeats the main frequencies (see above example with F1 and F2)

```
---/ Alternative Frequencies \-------------------------------------------------

AF Method A:

AF Method B:
List  1 = [#11, 106,6 MHz] 91,6 MHz, 106,6 MHz; 100,8 MHz, 106,6 MHz; 103,7 MHz, 106,6 MHz; 106,2 MHz, 106,6 MHz; 106,6 MHz, 107,6 MHz
List  2 = [#11, 106,2 MHz] 106,2 MHz, 106,6 MHz; 103,7 MHz, 106,2 MHz; 105,3 MHz, 106,2 MHz; 106,2 MHz, 107,6 MHz; 106,2 MHz, 101,9 MHz (RV)
List  3 = [#17, 103,7 MHz] 103,7 MHz, 106,6 MHz; 103,7 MHz, 106,2 MHz; 91,6 MHz, 103,7 MHz; 98,4 MHz, 103,7 MHz; 100,8 MHz, 103,7 MHz; 103,4 MHz, 103,7 MHz; 103,7 MHz, 105,3 MHz; 103,7 MHz, 107,6 MHz; 103,7 MHz, 101,9 MHz (RV)
List  4 = [#11, 91,6 MHz] 91,6 MHz, 106,6 MHz; 91,6 MHz, 103,7 MHz; 91,6 MHz, 103,4 MHz; 91,6 MHz, 105,3 MHz; 91,6 MHz, 107,6 MHz
List  5 = [#13, 105,3 MHz] 105,3 MHz, 106,2 MHz; 103,7 MHz, 105,3 MHz; 91,6 MHz, 105,3 MHz; 103,4 MHz, 105,3 MHz; 105,3 MHz, 107,6 MHz; 105,3 MHz, 101,9 MHz (RV)
```

windytan, 2023-06-14:
That's beautiful!

By sorting I mean, we shouldn't necessarily order the frequencies from smallest number to largest ourselves if the station is sending them in a different order. Maybe they are sent in a specific order for a reason - say, they could be closest or more powerful stations first.

So instead of this
```json
"regional_variants":[94300,96000,96900,97900,105800,107800]
```

We could leave them in the original order:
```json
"regional_variants":[94300,96000,97900,96900,107800,105800]
```

andimik, 2023-06-14:
Good question.

Can you repeat that without sorting it by frequency and test if you get the same output with `sort | uniq` like in the example?

andimik, 2023-06-14:
Could you share the output? Don't exactly know what you mean.

windytan, 2023-06-14:
Left: A numerical sorting algorithm was applied (the same text as above). Right: no sorting, frequencies are in the order they were received
[image]

andimik, 2023-06-14:
This is good. I feared that the not-sorting leads to multiple `tuned_frequency` lines for the same frequency.

So, I can say, the right column in your screenshot is fine for us.

closed 2023-06-14 by windytan

andimik, 2023-06-14:
Just one hint:

The normal output is printed in one single line.

```
$ cat /dev/ttyUSB0 | redsea -i tef -E | grep alt
{"pi":"0xA540","group":"0A","ps":"KAERNTEN","alt_frequencies_b":{"tuned_frequency":96100,"same_programme":[95700,101100,102100,104300,104900,107400]},"bler":0,"di":{"stereo":true},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA540","group":"0A","alt_frequencies_b":{"tuned_frequency":101100,"same_programme":[95700,96100,102100,104300,104900,107400]},"bler":0,"di":{"artificial_head":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA540","group":"0A","alt_frequencies_b":{"tuned_frequency":95700,"same_programme":[96100,101100,102100,104300,104900,107400]},"bler":0,"di":{"compressed":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
```
But will have additional new lines and the word `null` when you use it together with `jq`

```
$ cat /dev/ttyUSB0 | redsea -i tef -E | jq -r ".alt_frequencies_b.tuned_frequency"
null
null
null
null
null
null
null
null
null
null
104300
null
null
null
null
null
null
null
null
null
102100
null
null
null
null
null
null
null
null
null
null
```

```
$ cat /dev/ttyUSB0 | redsea -i tef -E | jq -r ".alt_frequencies_b.same_programme"
null
null
null
null
null
null
null
null
null
[
  95700,
  96100,
  101100,
  102100,
  104300,
  107400
]
null
null
null
null
null
null
null
null
null
[
  95700,
  96100,
  101100,
  102100,
  104300,
  104900
]
```

windytan, 2023-06-15:
jq has the option `-c` (compact output) that prints everything on one line. It prints `null` if the line did not contain the requested key. But I found this: https://blog.nem.ec/code-snippets/jq-ignore-nulls/

So your command could become something along these lines: `cat /dev/ttyUSB0 | redsea -i tef -E | jq -cr ".alt_frequencies_b.same_programme | select( . != null )"`

Or you could use `grep` before `jq`: `cat /dev/ttyUSB0 | redsea -i tef -E | grep same_programme | jq -cr ".alt_frequencies_b.same_programme"`

commit c9546250 (2023-06-14): Add support for AF Method B (fixes #88)

Redsea now decodes Alternative Frequencies transmitted with Method B.
This is used to transmit any number of different AF lists associated
with different tuned frequencies. Some of them may be transmitting
programme that varies region-by-region at different times of the day.

The JSON field alt_kilohertz was renamed to alt_frequencies_a. For
Method B there is a new field called alt_frequencies_b; it contains
frequencies for "same_programme" and "regional_variants" as well as
the tuned_frequency they're associated with.