windytan/redsea #89: Input from TEF6686 tuner

andimik, 2023-05-15:
Several radio freaks have recently bought a tuner with TEF6686 chip inside which can be connected to USB. See https://www.pe5pvb.nl/tef6686/

Its output (ttyUSB0) is in the following format:

```
PA540
R0549000000000F
Ss66.5,2,35
```
the line starting with P contains the PI-Code (here A540)
the line starting with R consists of RDS blocks B (`rdsB`), C (`rdsC`) and D (`rdsD`) and a RDS error (`rdsErr`) information
the line starting with S shows the signal strength.

Do you think it's possible to implement this format in redsea? I tried with `awk`, but it seems to be too tricky for me.

For example the following log
[a201_ttyUSB.txt](https://github.com/windytan/redsea/files/11482218/a201_ttyUSB.txt)
was filtered by RDS error = 0 (maybe other values are ok as well)
[a201.spy.txt](https://github.com/windytan/redsea/files/11482233/a201.spy.txt)

[image]

Specification of the chip:
https://github.com/makserge/tef6686_radio/blob/master/User_Manual_TEF6686.pdf

This is the source code which creates the format:

https://github.com/NoobishSVK/TEF6686-remastered/blob/330e4d591e0b09f664e36671db0b36627dc361f8/TEF6686_remastered.ino#L2251

windytan, 2023-05-17:
I took a look at it. The serial data looks quite odd - seems like the same group is always repeated four times, each with a different value of rdsErr. It would produce this:
```
A201 2010 4E91 ----
A201 2010 4E91 6368
A201 ---- ---- ----
A201 8003 ---- ----
A201 8003 C323 9AB3
A201 ---- ---- ----
A201 0018 ---- ----
A201 0018 F935 ----
A201 0018 F935 ----
A201 0018 F935 2020
...
```
But maybe it could work if only the ones with rdsErr=0 are accepted...

andimik, 2023-05-17:
Let me first try to modify the firmware for the device. So put this on hold for the moment.

I'm aware that the same group is repeated several times.

windytan, 2023-05-17:
Sure. It's practically done though, just missing some test cases :)

andimik, 2023-05-31:
The firmware has been modified and the above mentioned bug was fixed. The following log from `/dev/ttyUSB0` should be now ok.

[a502_serial.txt](https://github.com/windytan/redsea/files/11617772/a502_serial.txt)
[a213_serial.txt](https://github.com/windytan/redsea/files/11617773/a213_serial.txt)
[a203_serial.txt](https://github.com/windytan/redsea/files/11617774/a203_serial.txt)
[a201_serial.txt](https://github.com/windytan/redsea/files/11617775/a201_serial.txt)

The line with leading R and hex is the correct one, of course.

andimik, 2023-06-03:
R line has been fixed. Should now look like this:

[a502_ttyusb0.txt](https://github.com/windytan/redsea/files/11641975/a502_ttyusb0.txt)
[a203_ttyusb0.txt](https://github.com/windytan/redsea/files/11641978/a203_ttyusb0.txt)
[a201_ttyusb0.txt](https://github.com/windytan/redsea/files/11641979/a201_ttyusb0.txt)

I have tried with `sed`, but I have no idea how to extract the PI-Code from the P line (which is another line) so I add it manually ...

```
$ cat ./a203_ttyusb0.txt | sed -e '/.?/d' -e '/^S/d' -e  's/R//g' -e 's/..$//' -e '/^P/d' -e 's/^//' -e 's/^/a203/' | redsea -Eh | head -10
{"pi":"0xA203","group":"0A","bler":0,"di":{"compressed":false},"is_music":true,"prog_type":"Pop music","ta":true,"tp":true}
{"pi":"0xA203","group":"0A","bler":0,"di":{"artificial_head":false},"is_music":true,"prog_type":"Pop music","ta":true,"tp":true}
{"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA902","prog_type":"No PTY","ta":true,"tp":false},"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":0,"di":{"stereo":true},"is_music":true,"prog_type":"Pop music","ta":true,"tp":true}
{"pi":"0xA203","group":"2A","bler":0,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"8A","bler":0,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":0,"di":{"dynamic_pty":false},"is_music":true,"prog_type":"Pop music","ta":true,"tp":true}
{"pi":"0xA203","group":"8A","bler":0,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":0,"di":{"compressed":false},"is_music":true,"prog_type":"Pop music","ta":true,"tp":true}
{"pi":"0xA203","group":"8A","bler":0,"prog_type":"Pop music","tp":true}

```

```
$ cat ./a203_ttyusb0.txt | sed -e '/.?/d' -e '/^R/d' -e '/^S/d' -e  's/P//g'
a203
a203
a203
a203
a203
a203
a203
a203
a203
```

```
$ cat ./a203_ttyusb0.txt | sed -e '/.?/d' -e '/^S/d' -e  's/R//g' -e 's/..$//' -e '/^P/d' -e 's/^//' -e 's/^/a203/' | head -10
a203055911784f45
a203055a69602033
a203e54d0001a902
a203055f0e782020
a203255f20202020
a2038542c3239ab3
a203055815782020
a20385406021a201
a20305596c784f45
a2038542c3239ab3
```

windytan, 2023-06-04:
Can you check out the `tef6686` branch and see if it works all right?

windytan, 2023-06-04:
It's a new command-line option for input types: `redsea -i tef`

andimik, 2023-06-04:
Great! Confirmed, this works!

```
$ cat /dev/ttyUSB0 | ./redsea -E -i tef
{"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA502","kilohertz":95500,"tp":false},"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":0,"di":{"artificial_head":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"8A","bler":0,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":0,"di":{"stereo":true},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA502","prog_type":"No PTY","ta":true,"tp":false},"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":0,"di":{"dynamic_pty":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"2A","bler":0,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"8A","bler":0,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","alt_kilohertz":[88600,88800,90400,102500,103300],"bler":0,"di":{"compressed":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"2A","bler":0,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA213","tp":false},"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"Pop music","tmc":{"system_info":{"is_encrypted":false,"is_on_alt_freqs":true,"location_table":1,"scope":["national"]}},"tp":true}
{"pi":"0xA203","group":"8A","bler":0,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"0A","bler":0,"di":{"artificial_head":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA203","group":"2A","bler":0,"prog_type":"Pop music","tp":true}
{"pi":"0xA203","group":"14A","bler":0,"other_network":{"pi":"0xA213","tp":false},"prog_type":"Pop music","tp":true}
```

closed 2023-06-04 by andimik

andimik, 2024-02-05:
The output format has slightly changed in recent firmware version for several reasons. The only difference is, that in the "R"-Line now also the PI-Code is inserted.

Please note, that the old firmware might be still used widely.

```
$ cat /dev/ttyUSB0 
PA540
RA540054A52A84E5410
6

PA540
RA540255672204E6F10
40
RA540255672204E6F10
6

PA540
RA540054FED92454E00
PA540
RA540054892A8414E00
236
```
[antenne.txt](https://github.com/windytan/redsea/files/14171540/antenne.txt)

```
$ cat /tmp/antenne.txt | grep R | sed -e 's/R//g'
A5402553202D205400
A5402553202D205400
A540054F5288454E01
A5400548ED524B4100
A5400548ED524B4100
A54025546F6C642000
A54005495256455200
A54005495256455200
A540054A52884E5400
A540054A52884E5400
A5402555596F752000
```

```
$ cat /tmp/antenne.txt | grep R | sed -e 's/R//g' | redsea -i hex -E
{"pi":"0xA540","group":"2A","bler":0,"prog_type":"Pop music","tp":true}
{"pi":"0xA540","group":"2A","bler":0,"prog_type":"Pop music","tp":true}
{"pi":"0xA540","group":"0A","bler":0,"di":{"stereo":true},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA540","group":"0A","bler":0,"di":{"dynamic_pty":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA540","group":"0A","bler":0,"di":{"dynamic_pty":false},"is_music":true,"prog_type":"Pop music","ta":false,"tp":true}
{"pi":"0xA540","group":"2A","bler":0,"prog_type":"Pop music","tp":true}
```

So let me propose to add the new format, too, but keep compatibility to the old one (by checking the length of the "R"-Line).