windytan/redsea #90: Investigate migration to CMake

windytan, 2023-05-24:
Possibility to migrate from autoconf to CMake should be investigated. It would simplify the discovery of external dependencies and creating test cases, and also make the build system more accessible to developers.

I have a working CMakeLists.txt for:
* macOS

Remains to be tested:
* Linux (at least Ubuntu)
* Windows (WSL? MinGW?)

michal2229, 2023-05-26:
Please take a look at meson ( @https://mesonbuild.com ) - it's getting quite popular in FOSS world and I think it is pretty much justified. I used it in a few projects and I find it way easier and more readable/maintainable than CMake.

windytan, 2023-05-27:
Thanks! I'll check out the alternatives, too. Meson is a new one for me. The biggest plus side of good old CMake would be it's so widely used in C++ projects nowadays, and easily available for many platforms.

closed 2024-06-14 by windytan