windytan/redsea #92: ASCII Binary file not working

andimik, 2023-07-07:
According to the readme, the switch `-i bits` resp `--input bits` will load ASCII files with 1 and 0. But I don't get any output?

```
andreas@E535:~$ cat ~/Downloads/5218_binaer.txt | redsea --input bits
andreas@E535:~$ 

```
The file itself contains all 4 blocks, 0x5218 is `0101001000011000`

```
$ more ~/Downloads/5218_binaer.txt | grep 0101001000011000
0101001000011000000001010010110011100000110011010101001001010100
```

[image]

[5218_binaer.txt](https://github.com/windytan/redsea/files/11976761/5218_binaer.txt)

In the source code I also found the still existing switch `-b`

```
case 'b': // For backwards compatibility
```
But none of the methods is working here?

windytan, 2023-07-07:
Hmm. Could there be something up with the input data source? It seems as though the PI is not repeating every 104 bits but instead every 64 bits:
```
$ grep --only-matching --byte-offset '0101001000011000' ~/Downloads/5218_binaer.txt | head
0:0101001000011000
64:0101001000011000
128:0101001000011000
192:0101001000011000
256:0101001000011000
320:0101001000011000
384:0101001000011000
448:0101001000011000
512:0101001000011000
576:0101001000011000
```

windytan, 2023-07-07:
Maybe the 10-bit checkwords have been stripped out? That would make the group length to be 64 (40 bits missing from every group). Redsea can't decode binary without the correct checkwords, but you can first transform the data into the RDS Spy hex format, where they are not needed.

andimik, 2023-07-07:
Well, it's an old file in my archive. I will just convert the 4x 16 bits (which is 64 bits) to hex. Seems there is no additional overhead. That's why the PI is repeated after 64 bits. 

Unfortunately I don't know which tool has created that file.

You can close it, if you want.

closed 2023-07-07 by windytan

andimik, 2023-07-07:
But thank you for the `grep` parameter!

windytan, 2023-07-07:
No probs! Glad I could help