windytan/redsea #93: Input from soundcard

andimik, 2023-07-21:
In https://github.com/windytan/redsea/wiki/Use-cases#decoding-mpx-from-a-file-or-via-sound-card there is a howto when I want to use my 192kHz soundcard to feed it into redsea

But in my case, this did not work as everything above 48kHz was filtered out, see screenshot.

[image]

```
$ rec -t .s16 -r 171000 -c 2 - | redsea
rec WARN alsa: can't encode 0-bit Unknown or not applicable

Input File     : 'default' (alsa)
Channels       : 2
Sample Rate    : 171000
Precision      : 16-bit
Sample Encoding: 16-bit Signed Integer PCM

In:0.00% 00:00:08.53 [00:00:00.00] Out:1.45M [      |==-   ]        Clip:0    ^C
Aborted.

```
I found out that I need to specify the below mentioned parameters and used `-q` which stands for quiet.

```
AUDIODEV=hw:2,0 AUDIODRIVER=alsa rec -t .s16 -r 171000 -c 1 - -q | redsea -E
```

```
$ AUDIODEV=hw:2,0 AUDIODRIVER=alsa rec -t .s16 -r 171000 -c 1 - -q | redsea -E
rec WARN formats: can't set sample rate 171000; using 176400
rec WARN formats: can't set 1 channels; using 2
{"pi":"0xA201","group":"14A","bler":0,"other_network":{"pi":"0xA213","kilohertz":98600,"tp":false},"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"is_encrypted":false,"is_on_alt_freqs":true,"location_table":1,"scope":["national"]}},"tp":false}
{"pi":"0xA201","group":"8A","bler":0,"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"0A","bler":0,"di":{"stereo":true},"is_music":true,"prog_type":"No PTY","ta":true,"tp":false}
{"pi":"0xA201","group":"2A","bler":0,"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"14A","bler":0,"other_network":{"pi":"0xA213","prog_type":"No PTY","ta":true,"tp":false},"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"0A","bler":0,"di":{"dynamic_pty":false},"is_music":true,"prog_type":"No PTY","ta":true,"tp":false}
```
But due to the hint in the output this might be better:

```
$ AUDIODEV=hw:2,0 AUDIODRIVER=alsa rec -t .s16 -r 192000 -c 1 - -q | redsea -E -r 192000
rec WARN formats: can't set 1 channels; using 2
{"pi":"0xA201","group":"2A","bler":4,"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"8A","bler":4,"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"0A","bler":4,"di":{"dynamic_pty":false},"is_music":true,"prog_type":"No PTY","ta":true,"tp":false}
{"pi":"0xA201","group":"14A","bler":4,"other_network":{"pi":"0xA502","kilohertz":97000,"tp":false},"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"0A","bler":4,"di":{"compressed":false},"is_music":true,"prog_type":"No PTY","ta":true,"tp":false}
{"pi":"0xA201","group":"8A","bler":4,"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"0A","bler":4,"di":{"artificial_head":false},"is_music":true,"prog_type":"No PTY","ta":true,"tp":false}
{"pi":"0xA201","group":"14A","bler":4,"other_network":{"pi":"0xA502","kilohertz":99600,"tp":false},"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"0A","ps":"  OE 1  ","alt_frequencies_b":{"tuned_frequency":87700,"same_programme":[90900,92800]},"bler":4,"di":{"stereo":true},"is_music":true,"prog_type":"No PTY","ta":true,"tp":false}
{"pi":"0xA201","group":"2A","bler":4,"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"8A","bler":0,"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"0A","bler":0,"di":{"dynamic_pty":false},"is_music":true,"prog_type":"No PTY","ta":true,"tp":false}
{"pi":"0xA201","group":"2A","bler":0,"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"14A","bler":0,"other_network":{"pi":"0xA502","kilohertz":95600,"tp":false},"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"8A","bler":0,"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"0A","bler":0,"di":{"compressed":false},"is_music":true,"prog_type":"No PTY","ta":true,"tp":false}
{"pi":"0xA201","group":"2A","bler":0,"prog_type":"No PTY","tp":false}
{"pi":"0xA201","group":"14A","bler":0,"other_network":{"pi":"0xA502","kilohertz":95900,"tp":false},"prog_type":"No PTY","tp":false}
```
[image]

windytan, 2023-07-22:
Soundcard stuff can be highly platform specific. I suspect some devices may even silently upsample without providing the requested bandwidth...

I actually opened the wiki so it should be editable now. Would you be willing to add a short explanation about this in the wiki? Just something simple about the alternative recording command.

andimik, 2023-07-22:
I forgot to mention, this is Ubuntu 64bit

closed 2024-06-12 by windytan