windytan/redsea #95: Input from file only works with parameter `-f`

andimik, 2023-08-24:
https://github.com/windytan/redsea/blob/67c76f5d23454a26c1063495f123b98475305809/src/redsea.cc#L33

This hint seems not to work, I get no RDS data.

The only way is the parameter `-f [FILE]`.

```
andreas@E535:~$ more Downloads/radiostetin.wav.s16 | redsea -r 192k
{"pi":"0x0000"}
andreas@E535:~$ redsea -r 192k < Downloads/radiostetin.wav.s16 | head -5
{"pi":"0x0000"}
andreas@E535:~$ redsea -r 192k -f Downloads/radiostetin.wav.s16 | head -5
{"pi":"0x34EB"}
{"pi":"0x34EB","group":"0A","di":{"dynamic_pty":false},"is_music":true,"prog_type":"No PTY","ta":false,"tp":true}
{"pi":"0x34EB","group":"0A","di":{"compressed":false},"is_music":true,"prog_type":"No PTY","ta":false,"tp":true}
{"pi":"0x34EB","group":"0A","di":{"artificial_head":false},"is_music":true,"prog_type":"No PTY","ta":false,"tp":true}
{"pi":"0x34EB","group":"0A","di":{"stereo":true},"is_music":true,"prog_type":"No PTY","ta":false,"tp":true}
```

[radiostetin.wav.zip](https://github.com/windytan/redsea/files/12433323/radiostetin.wav.zip)
(just a sample file, taken from a friend)

windytan, 2023-08-25:
Seems that although it has the extension, this file is not in the s16 format (raw 16-bit signed integer PCM), instead it is a 24-bit WAV file.

```
$ file radiostetin.wav.s16
radiostetin.wav.s16: RIFF (little-endian) data, WAVE audio, Microsoft PCM, 24 bit, mono 192000 Hz
```

Some WAV files can works as raw PCM as the 16-bit aligned RIFF headers gets interpreted as a burst of noise, and they even can be piped to redsea as such, but the 24-bit sample width becomes a problem here.

Raw PCM normally looks like this (unless by coníncidence the noise starts looking like a file header):

```
$ file gen-yksi-171k.s16
gen-yksi-171k.s16: data
```

andimik, 2023-08-25:
I have renamed it as I thought it might work then. The original extension of the file was `.wav`

windytan, 2023-08-29:
You can convert any audio file (well, almost) into the S16LE format with sox: `sox radiostetin.wav radiostetin.s16` (this format has no headers for metadata, so make sure of the correct number of channels and sample rate).

For now, I'm not planning to add WAV input via stdin, but if such a need arises this can very well be reopened.

closed 2023-08-29 by windytan

andimik, 2023-08-29:
`-f` seemes to convert 24bit into 16bit as well. So it's worth mentioning it in the wiki as well.

windytan, 2023-08-29:
Good idea! For the wiki change, note that it’s converted direct to float32 (this also happens to the raw 16-bit input)

  - https://github.com/windytan/redsea/blob/c6e6b47ac2c7a9aac9409483b00ca61cd6eb47bd/src/input.cc#L92

Sent from mobile