windytan/redsea #108: build ERROR: Dependency "iconv" not found

andimik, 2024-06-17:
On my raspberry, I cannot build the latest v1.0 with meson.

```
$ uname --all
Linux rpi3 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux
```

```
pi@rpi3:~/apps/redsea $ meson setup build && cd build && meson compile
The Meson build system
Version: 0.56.1
Source dir: /home/pi/apps/redsea
Build dir: /home/pi/apps/redsea/build
Build type: native build
Project name: redsea
Project version: 1.0-SNAPSHOT
C++ compiler for the host machine: c++ (gcc 8.3.0 "c++ (Raspbian 8.3.0-6+rpi1) 8.3.0")
C++ linker for the host machine: c++ ld.bfd 2.31.1
Host machine cpu family: arm
Host machine cpu: armv7l
Configuring config.h using configuration
Compiler for C++ supports arguments -Wno-unknown-pragmas: YES 
meson.build:17: WARNING: Consider using the built-in optimization level instead of using "-O3".
Found pkg-config: /usr/bin/pkg-config (0.29)
Found CMake: /usr/bin/cmake (3.16.3)
Run-time dependency iconv found: NO (tried pkgconfig and cmake)

meson.build:22:0: ERROR: Dependency "iconv" not found, tried pkgconfig and cmake

A full log can be found at /home/pi/apps/redsea/build/meson-logs/meson-log.txt
```

```
$ which iconv
/usr/bin/iconv
```

[meson-log.txt](https://github.com/user-attachments/files/15862476/meson-log.txt)

andimik, 2024-06-17:
https://github.com/mpv-player/mpv/issues/10906

Will this help?

windytan, 2024-06-17:
Thanks for investigating this! I think it's because some systems have iconv as a separate library and some have its functionality bundled inside libc. I'll need to try out some hacks around this. Quick fix could be to remove all mentions of iconv in meson.build and see if it just compiles anyway.

andimik, 2024-06-18:
Some progress, I can build it, but there is a segfault then when executing `redsea`

```
iconv = cc.find_library('libc', dirs: ['/usr/bin'], static: true)
```

Please note, I am **not** a programmer (have never heard the term `meson` before) and just found some hints on the internet. No idea if I am wrong.

```
pi@rpi3:~/apps/redsea/build $ gdb ./redsea
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

--Type <RET> for more, q to quit, c to continue without paging--
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./redsea...(no debugging symbols found)...done.
(gdb) r
Starting program: /home/pi/apps/redsea/build/redsea

Program received signal SIGSEGV, Segmentation fault.
0x0008242c in __libc_start_main ()
(gdb) bt
#0  0x0008242c in __libc_start_main ()
#1  0x00024fa4 in _start ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

windytan, 2024-06-18:
Hi, this file should work on your system: [meson.txt](https://github.com/user-attachments/files/15889777/meson.txt). Just rename it to meson.build and replace the original. (GitHub didn't like the file extension, hence rename)

I'll write up a more portable solution later, that will support all systems, probably based on [this](https://www.mail-archive.com/qemu-devel@nongnu.org/msg750731.html)

andimik, 2024-06-18:
No, there is a problem in line 7.25

```
pi@rpi3:~/apps/redsea $ meson setup build && cd build && meson compile
The Meson build system
Version: 0.56.1
Source dir: /home/pi/apps/redsea
Build dir: /home/pi/apps/redsea/build
Build type: native build

meson.build:7:25: ERROR: Expecting eof got id.
meson.project_version()) configure_file(output
                         ^

A full log can be found at /home/pi/apps/redsea/build/meson-logs/meson-log.txt
```

andimik, 2024-06-18:
I have changed to your `dev` branch and could successfully build it.

commit 568321cc (2024-06-19): Fix meson.build for iconv on debian buster (fixes #108)

On Debian Buster, iconv needs different treatment. This solution
is loosely based on how qemu does it. As a last resort we can use
meson's discovery logic for iconv.
closed 2024-06-18 by andimik