windytan/redsea #115: Cannot compile on MacOS: library 'liquid' not found

kermit112, 2024-07-23:
Trying to compile the latest releases (1.0 and 1.0.1) but it stops with this error message:
The Meson build system
Version: 1.5.0
Source dir: /Users/janpeuser/redsea_old
Build dir: /Users/janpeuser/redsea_old/build
Build type: native build
Project name: redsea
Project version: 1.0.1
C++ compiler for the host machine: c++ (clang 15.0.0 "Apple clang version 15.0.0 (clang-1500.3.9.4)")
C++ linker for the host machine: c++ ld64 1053.12
Host machine cpu family: x86_64
Host machine cpu: x86_64
Configuring config.h using configuration
Compiler for C++ supports arguments -Wno-unknown-pragmas: YES 
meson.build:29: WARNING: Consider using the built-in option for language standard version instead of using "-std=c++14".
Found pkg-config: YES (/usr/local/bin/pkg-config) 0.29.2
Run-time dependency sndfile found: YES 1.2.2
Run-time dependency nlohmann_json found: YES 3.11.3

meson.build:70:20: ERROR: C++ shared or static library 'liquid' not found

Liquid-DSP is installed via brew. Maybe I'm holding it wrong, but you have a hint for fixing this.

windytan, 2024-07-23:
Hi, can you try this command and see what it prints? `brew --prefix liquid-dsp`

windytan, 2024-07-23:
And was this any help? [Readme#Troubleshooting](https://github.com/windytan/redsea?tab=readme-ov-file#troubleshooting)

kermit112, 2024-07-25:
Cloned the newest changes from the repository and now the compile worked perfect. 
Sorry for the noob question, but compiled it completely and running redsea -v still outputs "redsea 0.21-SNAPSHOT by OH2EIQ". How do I link to the newest build?

windytan, 2024-07-25:
No worries, it's a good question and something I completely forgot to address in the new Readme. It's because the old version is still installed somewhere. Can you run `which redsea` to see where it is?

You can remove the old executable and use these steps to build & install 1.x. It will get installed under `/usr/local/bin`.

1. Remove the build directory
2. `meson setup build --prefix=/usr/local`
3. `cd build && meson install`

windytan, 2024-07-27:
I made some changes so that, going forward, deleting the old binary and setting the `--prefix` option won't be necessary on macOS, since it will always install to /usr/local.

kermit112, 2024-07-31:
Makes it more noob-proof for sure, so thank you for your effort! I didn't have enough time to play with it in the last days, but this hopefully gets better in the future...

windytan, 2024-09-23:
Closing since the original issue was resolved, if there's any further trouble with the version let me know