windytan/redsea #122: Unit test(s) are failing with Slackware 15.0

pyllyukko, 2024-12-30:
Ehlo.

I was trying to build and package the new 1.1 version of redsea and ran into trouble when I have Catch2 installed:

```
FAILED: redsea-test-mpx 
c++  -o redsea-test-mpx redsea-test-mpx.p/src_block_sync.cc.o redsea-test-mpx.p/src_channel.cc.o redsea-test-mpx.p/src_dsp_liquid_wrappers.cc.o redsea-test-mpx.p/src_dsp_subcarrier.cc.o redsea-test-mpx.p/src_groups.cc.o redsea-test-mpx.p/src_input.cc.o redsea-test-mpx.p/src_options.cc.o redsea-test-mpx.p/src_rdsstring.cc.o redsea-test-mpx.p/src_tables.cc.o redsea-test-mpx.p/src_tmc_csv.cc.o redsea-test-mpx.p/src_tmc_events_data.cc.o redsea-test-mpx.p/src_tmc_tmc.cc.o redsea-test-mpx.p/src_tmc_locationdb.cc.o redsea-test-mpx.p/src_util.cc.o redsea-test-mpx.p/test_mpx.cc.o -Wl,--as-needed -Wl,--no-undefined -Wl,-O1 -Wl,--start-group -lliquid /usr/lib64/libsndfile.so /usr/lib/libCatch2Main.a /usr/lib/libCatch2.a -Wl,--end-group
/usr/bin/ld: i386 architecture of input file `/usr/lib/libCatch2Main.a(catch_main.cpp.o)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `/usr/lib/libCatch2.a(catch_session.cpp.o)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `/usr/lib/libCatch2.a(catch_test_case_info.cpp.o)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `/usr/lib/libCatch2.a(catch_test_spec.cpp.o)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `/usr/lib/libCatch2.a(catch_tostring.cpp.o)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `/usr/lib/libCatch2.a(catch_totals.cpp.o)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `/usr/lib/libCatch2.a(catch_version.cpp.o)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `/usr/lib/libCatch2.a(catch_assertion_handler.cpp.o)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `/usr/lib/libCatch2.a(catch_case_insensitive_comparisons.cpp.o)' is incompatible with i386:x86-64 output

...

[endless lines of failure follows]
```

So apparently it's trying to link against the multilib 32-bit version of Catch2's lib (living in `/usr/lib` vs. `/usr/lib64`). If I manually run the following (with `/usr/lib64`), it will work:

```
c++  -o redsea-test-mpx redsea-test-mpx.p/src_block_sync.cc.o redsea-test-mpx.p/src_channel.cc.o redsea-test-mpx.p/src_dsp_liquid_wrappers.cc.o redsea-test-mpx.p/src_dsp_subcarrier.cc.o redsea-test-mpx.p/src_groups.cc.o redsea-test-mpx.p/src_input.cc.o redsea-test-mpx.p/src_options.cc.o redsea-test-mpx.p/src_rdsstring.cc.o redsea-test-mpx.p/src_tables.cc.o redsea-test-mpx.p/src_tmc_csv.cc.o redsea-test-mpx.p/src_tmc_events_data.cc.o redsea-test-mpx.p/src_tmc_tmc.cc.o redsea-test-mpx.p/src_tmc_locationdb.cc.o redsea-test-mpx.p/src_util.cc.o redsea-test-mpx.p/test_mpx.cc.o -Wl,--as-needed -Wl,--no-undefined -Wl,-O1 -Wl,--start-group -lliquid /usr/lib64/libsndfile.so /usr/lib64/libCatch2Main.a /usr/lib64/libCatch2.a -Wl,--end-group
```

Not entirely sure how this should be addressed and I had to disable the unit test(s) from `meson.build` completely (see [here](https://github.com/SlackBuildsOrg/slackbuilds/pull/9245/files#diff-3b952ce71908410edd042e8120cd021ed398d40ce0879805e5e0d0de503d8f69)).

windytan, 2025-01-01:
Thanks for maintaining the SlackBuild pyllyukko, ja hyvää uutta vuotta.

I don't have much experience with Slackware, but I could confirm just now that it also happens for redsea 1.0.1 SlackBuild, and here's some bullet-point research for where it comes from:

* it comes from `pkg-config`. Here's some further info from `build/meson-logs/meson-log.txt`:

```
env[PKG_CONFIG_PATH]: /usr/local/lib64/pkgconfig:/usr/local/share/pkgconfig:/usr/lib64/pkgconfig:/usr/share/pkgconfig
env[PKG_CONFIG_ALLOW_SYSTEM_LIBS]: 1
env[PKG_CONFIG]: /usr/bin/pkg-config
------------
Called: `/usr/bin/pkg-config --libs catch2-with-main` -> 0
stdout:
-L/usr/lib -lCatch2Main -lCatch2
```
* Just for comparison, for sndfile, pkg-config is returning the 64-bit directory as expected.
* It seems that this library directory is set in `/usr/share/pkgconfig/catch2-with-main.pc`.
* This file is ultimately generated by CMake from [catch2-with-main.pc.in](https://github.com/catchorg/Catch2/blob/devel/CMake/catch2-with-main.pc.in) during the Catch2 SlackBuild compilation
* the libdir comes from a variable called `@CMAKE_INSTALL_FULL_LIBDIR@`. Not sure yet what is the story of this variable - to be continued.

windytan, 2025-01-01:
Could be an issue with the [Catch2 SlackBuild script](https://slackbuilds.org/slackbuilds/15.0/development/Catch2/Catch2.SlackBuild) in COMPAT32 mode?

See, when I build the Catch2 SlackBuild this happens that looks unintentional:

1. The build for 64-bit version begins.
2. The 64-bit version gets installed to the /tmp directory, along with this pkgconfig file:
```
grep libdir= /tmp/*/package-Catch2/usr/share/pkgconfig/catch2-with-main.pc
libdir=/usr/lib64
```
3. Now the Catch2 SlackBuild script notices that we are COMPAT32.
4. It starts to build the 32-bit version.
5. 32-bit version [gets installed](https://github.com/SlackBuildsOrg/slackbuilds/commit/19fc7bf6ceef79bc77baaf1da801d3c17019b836#diff-d4a59e52afd0754b633a3090cfb683c4e4ccf06b628da2790c5c951e9fd39ad7R114) to /tmp but the pkgconfig file gets overwritten since it has the same path for both 32 and 64-bit versions:
```
grep libdir= /tmp/*/package-Catch2/usr/share/pkgconfig/catch2-with-main.pc
libdir=/usr/lib
```
Like I said, it's the first time I used Slackware so you might want to take that with a grain of salt.

pyllyukko, 2025-01-01:
Hyvää uuttavuotta!

You are absolutely right. Should've done some more debugging before reporting this. Thanks for doing that for me :D I made a PR to SlackBuilds.org to get this sorted out.

With the referenced modification to the SlackBuild the unit tests succeed as they should.

Thanks!

closed 2025-01-01 by pyllyukko