windytan/redsea #133: Problems with ARI on 57kHz

andimik, 2025-06-03:
A German user has found out that a station in Salzburg (the town where Mozart was born) transmits the outdated ARI system (which was switched off in the 80ies) on 57 kHz.

In the screenshot, you will see a very strong but thin carrier between the RDS carriers.

https://en.wikipedia.org/wiki/Autofahrer-Rundfunk-Informationssystem
https://fr.wikipedia.org/wiki/Autofahrer-Rundfunk-Informationssystem
https://de.wikipedia.org/wiki/Autofahrer-Rundfunk-Information

German technical description at 
https://web.archive.org/web/20070727171248/http://www.ukwtv.de/artikel/technik/ARI-Technik.pdf

[image]

It seems that `redsea` has difficulties to detect more than the PI-Code. I hope this mpx recording is good enough, because it was captured from some distance and then even extracted from an SDR# (SDRsharp) recording.

```
cat ./mpx_welle_salzburg.wav | redsea -r 192000 -p
{"pi":"0xA851","group":"2A","tp":true,"prog_type":"Pop music","rt_ab":"A"}
{"pi":"0xA851"}
{"pi":"0xA851"}
{"pi":"0xA851","group":"0A","tp":true,"prog_type":"Pop music","di":{"stereo":true},"ta":false,"is_music":true}
{"pi":"0xA851"}
{"pi":"0xA851","group":"0A","tp":true,"prog_type":"Pop music","di":{"stereo":true},"ta":false,"is_music":true}
{"pi":"0xA851"}
{"pi":"0xA851","group":"2A","tp":true,"prog_type":"Pop music"}
{"pi":"0xA851"}
```

Is there anything we could improve to be able to detect RDS even when ARI is transmitted in parallel?

andimik, 2025-06-03:
Therefore, I'll close it.

vladisslav2011, 2025-12-04:
Hello.
This may be fixed by using complex band-pass filter instead of low-pass filter to extract RDS subcarrier from the MPX.
Complex band-pass will suppress the center frequency at 57kHz and keep sidebands. This works very well as RDS subcarrier does not have any energy at the center frequency.
For example, my Gqrx fork decodes RDS from the original IQ file just fine. It is able to decode even when using a low-pass filter but with reduced performance and lower SNR.
RDS Spy fails even when fed by properly filtered MPX from patched Gqrx...

[image]