windytan/redsea #136: Output sample number to JSON

zouppen, 2025-08-11:
I'm processing pre-recorded files and the sample number or relative seconds from the beginning of data would be useful for mapping the RDS events to timestamps.

For example:

```json
{"pi": "0x62A1", "group": "2A", "tp": true, "prog_type": "Light classical", "rx_i": 123456}
```

Where `123456` would be the sample number when the decoding was made. It doesn't need to be sample perfect in my use case. But of course extra epicness is guaranteed if it points to the exact start-of-frame sample number.

zouppen, 2025-08-11:
I use recordings which are in raw sample format (16-bit little-endian). Using hex-encoded format like in your example would unfortunately lead to 10 times larger input files which are no longer practical. I do full FM band decoding for radio propagation analysis.

You're probably correct, the exact sample accuracy would be probably difficult to achieve without extensive backtracking. But as I mentioned the exact sample accuracy isn't required in my case. Having the possibility map the "anomaly" to a timestamp and then to the original baseband signal timestamp is what I'm trying to achieve.

andimik, 2025-08-11:
Maybe you could have a look at https://github.com/andimik/fmlist_scan

This is a 24/7 analysis with RTL-SDR sticks logging all the time, for FM (incl. Redsea) and for DAB (which is not available in your country, but in Estonia).

If you are an FMLIST member, you can analyse them online.

[image]

Like in this log from another user today, he has received SporadicE from Italy on 93.1 MHz. It even was stronger then another German (but weak) station.

[image]

https://www.fmlist.org/urds/urdslister_fm_rds.php?id=22552

On your own Linux PC or Raspberry you can even check all logs and analyse all RDS files.

windytan, 2025-08-11:
It could be called `--relative-timestamp`?

zouppen, 2025-08-16:
Thanks! Converting it to seconds during printing (multiplying it with constant of 1/171000) would probably solve the issue in my case, since I'm interested about seconds, not exacet sample numbers. But of course, that's just my use case, not against other ways to format it.

windytan, 2025-09-06:
Naming is hard :) It's `--time-from-start`.

windytan, 2025-09-16:
Seems to work nicely for placing some overlays :) It's in the current master snapshot and will be in 1.3 once I figure out the Raspberry Pi issues. The timestamp is explained [here](https://github.com/windytan/redsea/wiki/Time-and-timestamps#timestamp-since-file-start)

[image]